MiSTer-devel/SNES_MiSTer #13: Component analog video not working

frankzzz, 2018-12-11:
Using SNES_20181207.rbf with component cables through IO board to my Philips CRT. Other cores like Genesis work fine. Tried flipping the 'SOG' switch on the IO board but that does not help.

pic:
[https://media.discordapp.net/attachments/497146208297680907/521533021959094273/JPEG_20181209_224533.jpg?width=500&height=669](url)

kow, 2018-12-13:
I experienced the same issue, appears to have been fixed in SNES_20181213.rbf

frankzzz, 2018-12-16:
I updated to the latest MiSTer core and SNES_20181213.rbf, still having the same issue

frankzzz, 2018-12-20:
Re-tested with SNES_20181219.rbf ..........same issue

sorgelig, 2018-12-20:
It's working. 
It's just your TV doesn't support the video from SNES core.

frankzzz, 2018-12-20:
The same TV works with a 2-chip original model SNES and a 1-chip SNES mini with HD Retrovision cables.....I realize the SNES core is not the same as a real SNES, but why is the sync signal so different from real SNES hardware where my TV won't accept it? Will this ever change? (I'm assuming its a sync problem, since the colors show up fine)

Thank you for responding @sorgelig , your work on the MiSTer project is very much appreciated. The other cores work fine on my TV so I'm very much a happy MiSTer user :smile:

sorgelig, 2018-12-20:
I don't know why your TV doesn't accept the signal.
SNES core is pretty much cycle accurate otherwise many games won't work. 
Try with scandoubler enable and disabled.

paulb-nl, 2018-12-21:
It seems Hsync is only 11 dots (2us) in SNES core. It should be 25 dots (4.7us) which is same as NES.  Maybe this is the problem.

sorgelig, 2018-12-22:
Ok. i will check it.

frankzzz, 2018-12-23:
I tried enabling scandoubler, that made the picture go further out of sync...
..HOWEVER..
...I just tried SNES_Test_20181222.rbf which was compiled by Soltan_G42 on the Classic Gaming discord server and it works! Not sure which commit fixed it, but I'm guessing the next official release it will be fixed as well