MiSTer-devel/SNES_MiSTer #14: Megaman X2 graphical glitch

dukesilverfan, 2018-12-15:
There is a small graphical glitch with release 20181213, where the number '2' is missing from the intro screen when loading Megaman X2. I have not tested the rest of the game to see if there are other glitches, but I confirmed that Megaman X3 shows the 3 on the intro screen, and they are both Cx4 games.

This is by no means game-breaking, but this glitch may affect other games in different ways. 
[image]
[image]

sorgelig, 2018-12-15:
this will be fixed along with with boxes.

dukesilverfan, 2018-12-20:
Confirmed fix with Release 20181219.

closed 2018-12-20 by dukesilverfan