MiSTer-devel/SNES_MiSTer #20: Ken Griffey Jr. Presents Major League Baseball

RetroDriven, 2019-01-02:
When selecting the option "Season" for the game mode it doesn't work. You will see a weird graphical glitch and the game either freezes up or resets.

James-F2, 2019-01-26:
Confirmed.

James-F2, 2019-01-30:
Fixed

James-F2, 2019-03-04:
Looks like it was fixed.
Some commit from March 3rd or 4th fixed this bug.
Needs confirmation by another person and close.

James-F2, 2019-03-05:
Confirmed to be fixed by others.

Main MiSTer had too be updated too, probably because of this:
> Report image size 0 if no save available

RetroDriven, 2019-03-05:
I'm running the latest Mister main and SNES Core but I'm still having the same issue. Maybe it's my Rom?

James-F2, 2019-03-05:
Sorry forgot to mention that it is with March 4th commits that I compiled, they will be included with the next release.

James-F2, 2019-03-09:
Works perfectly with latest release.

closed 2019-03-24 by sorgelig