MiSTer-devel/SNES_MiSTer #26: Uniracers, 2P & VS  mode, second player bug

James-F2, 2019-01-26:
When playing a 2 player game, the second player is hovering above the track, he should be on the track.

James-F2, 2019-02-25:
More info;
The bottom player (player 2) nose is always facing left even though he's driving to the right.
When both players on a single screen, the bottom player disappears from the bottom screen.

Something even stranger:
When the bottom player comes close to the top player from the left, he suddenly teleports to the right of the top player.
Moreover, there is a part of the upper screen where the bottom player can't be in where I ..
[image]

It seems like a DMA problem with Object Attribute Memory.
As explained here:
https://youtu.be/K7gWmdgXPgk?t=12m47s

srg320, 2019-02-25:
You are right and this problem is already being [discussed](https://forums.nesdev.com/viewtopic.php?f=12&t=18447)
This will be fixed in future releases.

paulb-nl, 2019-02-25:
Could you test this build? https://github.com/paulb-nl/SNES_MiSTer/tree/oam/releases

I have made big changes to sprite evaluation to be closer to a real SNES so it needs testing to make sure there are no regressions.

James-F2, 2019-02-25:
Will do, right now.

James-F2, 2019-02-25:
I have tested 20+ game including Uniracers.
I see no glitches at all.

paulb-nl, 2019-02-26:
Thanks for testing James. Good to hear there are no issues.

James-F2, 2019-02-26:
@srg320 mentioned on the nesdev thread that he is reworking the PPU sprite handling too,, so maybe he has a fixed PPU version too?

srg320, 2019-02-26:
Of course, but later, now I'm busy SA1.

James-F2, 2019-02-27:
This bug has been fixed with paulb_nl request.
https://github.com/MiSTer-devel/SNES_MiSTer/pull/44

sorgelig, 2019-02-27:
ok. then I close it.

closed 2019-02-27 by sorgelig