MiSTer-devel/SNES_MiSTer #28: Soul Blazer, Illusion of Gaia: in game save not working

michaelshmitty, 2019-01-28:
Games:
* Soul Blazer
* Illusion of Gaia

Reproduce:
1. Start new game.
2. Create new journey, play a bit and save in game.
3. Reset game to find no record of your journey.

Interestingly in the third game in this series by the same producers, Terranigma, the in game save works fine. Perhaps the former two games used an older method of using in game saves? Just guessing here.

sorgelig, 2019-01-28:
Don't you forget to press the "Save backup RAM" in OSD before leave the game?
What you mean by "Reset game"?

James-F2, 2019-01-28:
Works fine here.

michaelshmitty, 2019-01-28:
> Don't you forget to press the "Save backup RAM" in OSD before leave the game?

I will try that. I didn't do that in Terranigma, Super Mario World, Yoshi's Island (thanks for the Super FX support by the way!) so I'm curious as to why for some games you have to "Save backup RAM" to keep the in game save data and some not.

> What you mean by "Reset game"?

I meant reset core from the OSD menu, my mistake. Or power cycle the MiSTer board.

James-F2, 2019-01-28:
All games need "Save backup RAM" to save, no exclusion.
The "Reset" you see in the core menu is a soft reset and will not discard your save, BUT once you change a rom or do a hard mister reset, it will be gone if you did not use the "Save backup RAM" function.

When Loading a rom, it does "Load backup RAM" automatically.

michaelshmitty, 2019-01-28:
@James-F2 Cool, thanks for that insight. I'll take some time to figure it all out. I'm assuming this is not an issue and rather a misunderstanding at my end. Cheers!

closed 2019-01-28 by michaelshmitty