MiSTer-devel/SNES_MiSTer #37: Super Fx Speed

Gospoder, 2019-02-24:
Star Fox on the original setting is very slow. Compare the current version with this [video] (https://www.youtube.com/watch?v=PpA0fKhiFO4). It may be the plot instruction. Another comparison can be made with the intro to Vortex.

sorgelig, 2019-02-24:
Original speed of StarFox is half speed of GSU. Probably on video they use normal GSU speed which is twice higher than StarFox uses. 
Turbo speed has no any artificial throttles, so it's faster than on that video.

cacophony1, 2019-02-25:
The left side of the video shows an actual retail/unmodified Starfox cart running on a real SNES.  It sounds like the MiSTer core is 5 seconds behind that video at about the 1 minute mark of the video.

sorgelig, 2019-02-25:
I will try to set a normal speed for GSU and see how it will match.

James-F2, 2019-02-25:
I remember redguy who did SFX and SA1 on the sd2snes had to tweak the cycle count on certain emulated components, but not changing the SuperFX master clock.

James-F2, 2019-03-04:
StarFox is MUCH closer to the original cartridge now with srg320 latest commit.

James-F2, 2019-03-04:
Verified. 
Timing is now spot on in StarFox ending.
Well Done!

Can be closed.

closed 2019-03-04 by sorgelig