MiSTer-devel/SNES_MiSTer #38: Chrono Trigger has small graphical glitches when a battle starts

the-importer, 2019-02-24:
It happens on my 4KTV and my LCD monitor. I tried different resolutions and Vsync options, but I still get the issue.

James-F2, 2019-02-25:
Only happens on HDMI, not VGA.
It is probably scaler related.

Save attached.
Go t into town and fight the creatures.
[Chrono Trigger (U) [!].zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/2900744/Chrono.Trigger.U.zip)

the-importer, 2019-02-25:
> Only happens on HDMI, not VGA.
> It is probably scaler related.
> 
> Save attached.
> Go up into Truce Canyon (mountain) and fight the creatures.
> [Chrono Trigger (U) [!].zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/2900744/Chrono.Trigger.U.zip)

Actually, my 2nd screen, that old LCD monitor that I was mentioning, is hooked via VGA and the issue is actually worst on it. Also, why would I fight the creature in your save game? Will is fix anything?

James-F2, 2019-02-25:
I only see this with HDMI.
Do you use the VGA output from the IO board?

The save file I posted because it takes 10 minutes till you actually fight monsters in this game, and you did not provide one.
It is for developers who want to test this bug quickly.

the-importer, 2019-02-25:
> I only see this with HDMI.
> Do you use the VGA output from the IO board?
> 
> The save file I posted because it takes 10 minutes till you actually fight monsters in this game, and you did not provide one.
> It is for developers who want to test this bug quickly.

Yes, I use VGA from the IO board. As for the your save data, I don't really need it. Right at the beginning, I can just go to the forest, fight the monsters there and see the glitch.

James-F2, 2019-02-25:
Again, the save is not for you but for developers who do not know where the forest is or played chrono trigger, and want to test the reported bug quickly.

Also, sometime this bug appears only after reset.

James-F2, 2019-02-25:
Do you have vga_scaler enabled in mister.ini?

the-importer, 2019-02-25:
> Do you have vga_scaler enabled in mister.ini?

I fiddled with it, didn't change anything. If I didn't know any better, it's almost like CT changes resolution when entering battle mode (similar to how how Chrono Cross switches from 240p to 480i in the menu).

James-F2, 2019-03-04:
I can see the "before fight" image flicker when loading a rom.
It's probably HDMI buffer related.

the-importer, 2019-03-04:
@sorgelig
And I see the issue with both VGA (on an old LCD monitor) and HDMI on my TV.

James-F2, 2019-03-04:
@the-importer 
Can you please post your mister.ini file, that would help solve the problem.

paulb-nl, 2019-03-14:
I don't see this issue with VGA to CRT or OSSC. 

The game does a lot of HDMA writes per frame to register 2130-2133. Interlaced is one of the things set with 2133. So I made a build with interlaced disabled in the Ascal scaler and the issue was gone.

Appearantly interlace is not enabled long enough for the analog output to be affected but the scaler reacts immediately.

Either the game enables interlace by accident or it is a bug in the SNES core.

James-F2, 2019-05-07:
If it is the game that write to 2133 then this issue can be closed.
@paulb-nl  can you confirm that it is indeed the game that does this?

sorgelig, 2019-05-07:
probably interlace mode can be locked at some point of screen to eliminate sporadic changes while screen drawing

paulb-nl, 2019-05-07:
@James-F2  I have checked with Mesen-S and the game enables interlace on scanline 8 and disables on scanline 9 for 2 frames in a row.

James-F2, 2019-05-07:
Thank you.
So it is obviously not a bug in the core or scaler.
Should it be fixed as a workaround quirk or ignored?

paulb-nl, 2019-05-07:
Well It's kind of a bug in the scaler because I don't think it should detect interlace when interlace is only enabled for 1 scanline of the frame.

Could probably be fixed if the scaler only detects interlace in Vblank or skip detection and send interlace enabled bit to the scaler.

sorgelig, 2019-05-07:
It's not only about scaler. SNES glue logic i wrote disables scandoubler if interlace is detected (scandoubler is not compatible with interlace). So it doesn't lock interlace mode either. So i need to update my code to not react on sporadic changes of interlace mode.

paulb-nl, 2019-05-14:
Latching the INTERLACE flag only on Vsync start seems to be working for both.
````verilog
if(~old_vsync & VSync) begin
	frame_hres <= tmp_hres | ~scandoubler;
	tmp_hres <= HIGH_RES;
	Interlace_l <= INTERLACE;
end
````

sorgelig, 2019-05-14:
i forgot about this issue :)
will add it.

James-F2, 2019-05-15:
Seems to be fixed in 20190515.
Thank you paulb-nl & sorgelig!

closed 2019-05-15 by sorgelig