MiSTer-devel/SNES_MiSTer #39: Super Mario RPG - Crash on 1st Level Up screen

Hewhoisred, 2019-02-25:
New with the SA1 chip addition the game is playable.  This bug will take about 10 minutes after starting a new game to replicate or use the attached sav.  Go kill one group of monsters and you'll level up.  At the first level up bonus screen, the game hard locks, with background music continuing.

[Super Mario RPG - save.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/2898699/Super.Mario.RPG.-.save.zip)

molnara, 2019-02-25:
I can confirm this is an issue.

Hewhoisred, 2019-02-25:
Further info:  Level up screen uses character conversion dma.
This was apparently a very difficult bug to squash on the SD2SNES FPGA, according to Redguy, one of it's developers.

Hewhoisred, 2019-03-03:
Fixed with the last update.  You can close.  Thanks very, very much!

closed 2019-03-03 by sorgelig