MiSTer-devel/SNES_MiSTer #40: Soul Blazer - music rhythm problem

James-F2, 2019-02-27:
When starting Soul Blazers you can hear the drums are losing rhythm and not sounding correct.
APU timing issue?

Example of the bug:
https://cdn.discordapp.com/attachments/446396142608056320/549803272924299265/video.mov

* Same sound bug exists on the SuperNT

sorgelig, 2019-02-27:
recording is too quiet - cannot hear the problem.
May be it's like Streets Of Rage 2 on Genesis where people reported that sound is off in player selection while it's originally off by musician.

James-F2, 2019-02-27:
Same bug on the SuperNT:
https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/issues/58

sorgelig, 2019-02-27:
And where people said it's also happens on original HW.
looks like baggy game itself.

James-F2, 2019-02-27:
It is not buggy on my 1Chip console, it sounds correct on my console.

James-F2, 2019-02-27:
Timing bug stops happening after loading the game several times on the MiSTer.
Maybe it is related to the APU state?
It seems like the game 'fixes' the APU state...

sorgelig, 2019-03-24:
Some games forget to initialize some registers at start and relay on current state which might be incorrect. So, strictly speaking it's bug of the game rather than core's bug.

James-F2, 2019-04-10:
Can be closed.

closed 2019-04-10 by James-F2