MiSTer-devel/SNES_MiSTer #47: Kirby Super Star intro slowdown

paulb-nl, 2019-03-03:
In the intro when Kirby is flying over the grass there is slowdown in some parts. Video: https://www.youtube.com/watch?v=Mc5dhkkBWdw

During the slowdown a blue bar flickers in the background just above the grass.
[image]

On real hardware there is no slowdown: https://youtu.be/tGt0QQkkY1c?t=112

It seems that the early SA-1 core on SD2SNES had the same problem: https://youtu.be/Jw5_Z3CE2SM?t=958

James-F2, 2019-03-04:
Using this SA-1 Speed Test rom:
https://github.com/VitorVilela7/SnesSpeedTest

The clock speeds look exactly like on the real hardware SA-1, but the game still stutters.

paulb-nl, 2019-03-22:
Awesome work by srg320. Works fine now

sorgelig, 2019-03-23:
TG3000 is fixed as well!

sorgelig, 2019-03-24:
I close it as fixed.

closed 2019-03-24 by sorgelig