MiSTer-devel/SNES_MiSTer #51: Analog audio: low volume digital buzzing sound in left speaker during silence

cr4zymanz0r, 2019-03-06:
Core 190303

Issue is present in analog audio output (via IO board), but not on the HDMI board.

Was very noticeable in Mega Man X 2. Also noticed in Final Fantasy 6. Did not extensively test for it in many games.

MordecaiWalfish, 2019-03-06:
I can confirm that the newest released snes core broke analog audio. I have been extensively testing the SNES core for issues for the past couple weeks, and 3.5mm analog audio was pristine before the recent change (SNES core update from March 3rd)

Now when games fade to silence, you can hear both faint static/noise at time and some times you can faintly hear some weird frequency modulation sounding noise (best i can describe it). 

Funny thing, is that the TGFX16 core had these EXACT same issues about a month ago when I first started using my MiSTer, but this was fixed in the TGFX16 core update that came out on Feb 16th.

SNES core has regressed to that exact issue, whatever it is. i have headphones connected directly to my MiSTer and you can hear this noise during the intro of some games after the logo jingles play, or in games like SF2, in-between rounds.

James-F2, 2019-03-06:
I too can hear some audio "whine" from analog audio with official March 3rd release.
But, I compiled March 5th to test, and it does not have this issue with analog audio,, something is wrong with official March 3rd build analog audio.

MordecaiWalfish, 2019-03-14:
This was fixed in the 3-9-19 core release for me.

cr4zymanz0r, 2019-03-27:
Tested again with the latest release and it seems to be resolved. Analog audio from MiSTer seems noisy in general, but is not obvious at normal listening volumes on speakers like it was on core 190303

closed 2019-03-27 by cr4zymanz0r