MiSTer-devel/SNES_MiSTer #53: Super Mario RPG - Stray pixel noise/crash

Hewhoisred, 2019-03-07:
Latest 20190308 release, there are minor stray pixels around sprites.  Also, the game crashes not long after you load the included save (see attached), usually after you go down the stairs.

Other user experience with the same sav:
"actually after the dialogs something weird happens
seems to sort of get stuck with mario able to jump but not move
yeah, something weird there"

"sometimes freezes up at the opening "sky" screen
a reset can fix it but loading in the save it will freeze up as Mario is falling in on the save block and Mario has some stray pixels around him"

[Super_Mario_RPG_-_Legend_of_the_Seven_Stars_USA.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/2943517/Super_Mario_RPG_-_Legend_of_the_Seven_Stars_USA.zip)

James-F2, 2019-03-08:
The compilation of 20190308 release is bad.
I have compiled the core and it does not have any issues.
Even without this save SMRPG freezes in the intro sequence with 20190308 release.

sorgelig, 2019-03-08:
due to large size and many processors this core got stability issue...

James-F2, 2019-03-08:
I have tried to compile 3 times,, and now I have this issue too.
Everything was okay on March 4th.

Maybe this one did it?
https://github.com/MiSTer-devel/SNES_MiSTer/commit/2d889bf315577e903888f6eccdfb3c48edf729a5

sorgelig, 2019-03-08:
When battery is off or on first start the SRAM has garbage. If just clear to zero as it was before, then Baseball game won't work for example.

sorgelig, 2019-03-08:
tried intros in Kirby and SMRPG - both work fine on both my DE10nanos with 0308 release. 
So it's definitely a metastability issue.

James-F2, 2019-03-08:
I have complied again with old snes.sv before ""//Thrash the BSRAM upon ROM loading"",, and it does not freeze now.

Did you watch the entire intro or just few seconds?
It freezes/crashes when mario jumps on the koopa's during gameplay sequence.

sorgelig, 2019-03-08:
Any change in code will produce other logic map, so it may fix the freezing or make it worse regardless which part of code you change. It has no relation to that commit.
Which application relays on data from RAM on power up? It's nonsense!

sorgelig, 2019-03-08:
one of my DE10 freezes on SMRPG demo if i turn off the fan. So i can test in this board.

James-F2, 2019-03-08:
SA1 game rely on the the battery powered sram on the cartridge for saves and as working ram,, so maybe the expect certain state?

sorgelig, 2019-03-08:
so, if the battery will off you have to throw away the cartridge? ;)

Hewhoisred, 2019-03-08:
I can confirm that even without a .sav file (or with a brand new one if the game auto generates one at first run), the game crashes on the third scene of the intro where Mario is jumping around avoiding monsters.

Hewhoisred, 2019-03-08:
All issues appear to be fixed with the re-release.  Intro finishes fine, no noise on sprites and no more crashes during gameplay.  Thanks for the hard work.

James-F2, 2019-03-08:
Kirby Super Star still crashes for me and other person with the 0308 re-release.

I have compiled with default optimization settings in Quartus and it doesn't seem to freeze anymore in Kirby.

James-F2, 2019-03-08:
StarFox (GSU) freezes when switching to Turbo speed with the re-release.

kousaka, 2019-03-08:
I'm getting graphical issues in multiple games with the original release and the re-release.

sorgelig, 2019-03-08:
try this build: https://mega.nz/#!1cZwQS5K!5HLse-2HVkn1KXClYSonku1re1esoUWSzGnSDmHgrYw

James-F2, 2019-03-08:
For me this build fixed all the reported problems on this issue.

Hewhoisred, 2019-03-08:
I tested SM RPG intro and Kirby Superstars intro with the megalink build.  No crashing, no visual glitches.

sorgelig, 2019-03-08:
good!

closed 2019-03-08 by sorgelig

kousaka, 2019-03-09:
I can confirm the new version has fixed everything also