MiSTer-devel/SNES_MiSTer #54: Secret of Mana sound bug (audio channel improperly cutting out)

MordecaiWalfish, 2019-03-08:
There is a sound issue in the sewers that travel between the 'southtown' and 'northtown' areas of Secret Of Mana, that does not occur on original SNES hardware.

[Secret of Mana (USA) [northtown save].zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/2946314/Secret.of.Mana.USA.northtown.save.zip)

Steps to re-create issue (save file attached):

Load the top savegame, exit the inn, go down the stairs, and enter the cafe in the south-central part of the town. Walk all the way through the inn, and into the sewers.

As soon as you enter the sewers, you will hear a 'static-sounding' sound effect that is supposed to be flowing water, and it will cut off immediately when another sound is played.

This static sound should play uninterrupted when you are in this section of the game, besides in a few momentary instances that I have outlined below. This has been verified on original SNES hardware.

If you go between screens in this area, you will hear the same thing in each area-- the water flowing sound effect will play as soon as you enter the screen and immediately cut off when another sound effect is playing.

When the main character is equipped with the whip, you can force this static noise to play by attacking, and while your attack is charging up to 100%, the water flowing/static sound effect can be heard playing again, and then it will instantly stop when the beep plays to indicate that you are charged.  It is easiest to do this if you clear the screen of enemies first, as any sound effect they make will interfere with this.

This has been verified on original SNES hardware, and can also be observed in the following youtube longplay for Secret of Mana:

Original sound can be heard here: https://youtu.be/dxiF3uxmfSU?t=5478

As you can hear in this example (which luckily has the main character also equipped with the whip), the sound should be pretty much constant after you walk into this area. 

The only time this sound cuts out on original SNES hardware is sometimes when magic is being cast: https://youtu.be/dxiF3uxmfSU?t=5731
and also very shortly when enemies are hit by a weapon (most noticeable with the whip it seems)

James-F2, 2019-03-09:
I confirm.
On the real console the waterfall sound never stops.
Easiest way to test is to start a new game and wait 30 seconds for the waterfall screen.

MordecaiWalfish, 2019-03-09:
I felt kind of dumb giving all those steps and that save file and then finding out that the very beginning does this too, but I left the more complicated method and save file in the post as well, in hopes that actually being able to control when the sound effect plays by using the whip may help debug this issue.

James-F2, 2019-04-10:
Seems to be fixed with latest April commits by srg320.

MordecaiWalfish, 2019-04-10:
Issue is resolved in all recent test builds; closing.

closed 2019-04-10 by MordecaiWalfish