MiSTer-devel/SNES_MiSTer #59: Mister FPGA - Marvelous - Mouhitotsu no takarajima 1.07

terminator2k2, 2019-03-22:
bug in marvelous , chapter 5  maps

[image]

should look like 

[image]

save file 

[Marvelous_-_1.07.sav.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/2998767/Marvelous_-_1.07.sav.zip)

using the save file follow this youtube to 1min 30 seconds

 https://www.youtube.com/watch?v=Q8rc7B0vu4U&t=96s

srg320, 2019-03-23:
What hardware was the second photo taken from?
Interesting, on this rom bug appears both on bsnes, snes9x and no&sns, but in Japanese and English by Tashi versions bug does not appear anywhere.

terminator2k2, 2019-03-23:
yes just noticed that this issue only appears on 1.07 version of the translation, just tried the other versions and they are fine, so ill close this...

closed 2019-03-23 by terminator2k2