MiSTer-devel/SNES_MiSTer #63: Killer Instinct Sound Glitches & Freezing

MordecaiWalfish, 2019-04-04:
Tested with latest official SNES release (3-3-19) and more recent builds compiled by James.

Steps to reproduce: 
Start single player game and play normally. The game will freeze and/or the audio will stutter in a loop. Sometimes sound will freeze entirely and gameplay will continue for a round or two until the whole game eventually freezes.

This seems to happen more commonly if you mash all the buttons while playing. Try setting turbo to all buttons and holding them down in-between rounds, etc and you will likely start a round with the sound glitching.

James-F2, 2019-04-12:
I confirm.

Sometimes it will have audio glitches or play different samples altogether.
Sometimes, the game will freeze during a fight.

srg320, 2019-04-12:
Yes, the problem is in SPC700, I am working on it now.

srg320, 2019-05-10:
Try the latest release.

MordecaiWalfish, 2019-05-11:
I tested with the latest release and the same issues occur. 
sound cut out before i even got to the character selection the first time i tried. reset and sound came back but then about 5 fights into single player mode the game froze

James-F2, 2019-05-11:
Yes this still happens, especially on a fresh boot of the core.

James-F2, 2019-06-10:
Seems to be fixed in 20190609 release.
I don't see SPC700 commits but I don't experience freezes or audio glitches anymore.
Maybe it was memory related outside the core?

Needs to be confirmed by at least one more person to close issue.

Hewhoisred, 2019-06-10:
Played about an hour, completed the game.  No issues that I could see.

srg320, 2019-06-10:
I also do not see the issue.

James-F2, 2019-06-10:
This is very strange.
I just checked previous releases from 2019.05.10 and they all have issues with this game, but not 20190609 release.
I don't see a particular commit that might have fixed this.

srg320, 2019-06-10:
@James-F2, this commit fafa721d55c33237c927f7690e771906987105cf

James-F2, 2019-06-10:
Ah, yes, I thought it was DSP1-4 special chip related, and not general signal processing.
Absolutely amazing work.

MordecaiWalfish, 2019-06-11:
I can confirm this issue is no longer occurring, neither the sound problem or freezing.

I played through for over an hour without any issues, fell asleep for 6 hours with the MiSTer running the demo loop and when I came back to it everything was still functioning properly.

Thanks for your time and attention in addressing this, and your continuously impressive work with the console cores, @srg320 . I dont know how tough it would be to address the turbografx16 issues, but there has been a long-standing graphics problem with that core, that affects some pretty big games like SF2:CE. Seeing that core come to parity with the current quality/stability of snes/nes/genesis cores would be a big step forward IMO.

I am closing this issue.

closed 2019-06-11 by MordecaiWalfish