MiSTer-devel/SNES_MiSTer #69: OAM behavior not accurate

paulb-nl, 2019-04-28:
Here is a video of my OAM test ROM on a real NTSC SNES: https://youtu.be/9aJ7ZsonZXk

ROM: https://drive.google.com/file/d/1foFV_RJbMUGWaQS_D-3J_E-XMz06Bilv/view?usp=sharing

Release 20190412 does not have accurate OAM behavior. With the Force blank bar there are no sprites on the next line. Also no sprites disappear when pressing button B.

A build with commits until 22 march is very accurate to the video.

This is what the test does:

- Add 128 sprites to OAM table. Every sprite in a row is offset by 4 so the first sprite is sprite 0 and second sprite is sprite 4. On the second row the first sprite is 1 and second sprite 5.
Writing to High OAM will affect 4 sprites (1 sprite from every row).

- Set IRQ (adjust with d-pad)

In IRQ:

- Enable Force blank
- Write OAM address $2102 (adjust with Y/A buttons)
- Disable force blank. Low brightness(write #$08 to $2100)
- Wait for some time (adjust with L/R shoulder buttons)
- Write OAM (#$55 to $2104) to hide sprites off-screen (Hold B button)
- Set full brightness (write #$0F to $2100)

These are the controls:
D-Pad: adjust IRQ position
L/R shoulder buttons: adjust delay between disable Force blank and write to OAM
B: Hold to write to OAM
Y/A: Decrease/Increase OAM address
Select: Reset OAM address to 0

In the video I start with pressing B and you can see 4 sprites disappear. Moving the force blank bar left and right does not change which sprites disappear because after disabling Force blank OBJ-Range continues sprite evaluation at the OAM address set in force blank

*There is a glitch with sprite evaluation in some conditions where much less sprites will be shown on the next line. This is visible when I move the forceblank bar left and right.

Then I press the L/R shoulder buttons to increase/decrease the delay before writing to $2104. The delay changes which sprites disappear.

At the last part I press the A button to increase the OAM address. The sprites are 16 bytes apart so every 8 button presses the sprites which disappear moves to the right. (Sprite evaluation continues at OAM address set in Force blank)

srg320, 2019-05-01:
@paulb-nl, thanks for your tests and especially thanks for video. Unfortunately I don't have real SNES and I compare only with PC emulators (mostly with bsnes). If you have other tests and video, then I would be very thankful.
I implemented your changes c0838f063b0d131afa50ffbd7efca70d996bfbdc in current code and hope now it is accurate. Please try [this ](https://yadi.sk/d/Qbm8zULcwSKOdg) PPU version.

paulb-nl, 2019-05-01:
Thanks. I have tried your changes. It is almost accurate but the next line of sprites beneath the force blank bar is constantly flickering.

Unfortunately Bsnes is not accurate at all with sprite evaluation. I hope Mesen-S will be accurate in the future.

paulb-nl, 2019-05-02:
@srg320 
The flickering sprites are caused by this part:
````vhdl
if OBJ_RANGE = '1' and FORCE_BLANK = '0' and DOT_CLKR_CE = '1' then
	OAM_ADDR <= std_logic_vector(unsigned(OAM_ADDR) + 2);
end if;
````

It is not aligned with H_CNT(0) so depending on which cycle forced blank is disabled sprite evaluation will use the wrong OAM address.

Previously it was like this:
````vhdl
if DOT_CLKR_CE = '1' and OBJ_RANGE = '1' and H_CNT(0) = '1' then
     OAM_ADDR <= std_logic_vector(unsigned(OAM_ADDR) + 4);
end if;
````

srg320, 2019-05-05:
H_CNT(0) select OAM data word of one sprite for reading, but if you think that read 32 bits then I will not argue.
With the [current ](https://yadi.sk/d/Qbm8zULcwSKOdg) version PPU the Uniracers, Star Ocean and HblankEmuTest work fine and your test looks like too.

paulb-nl, 2019-05-05:
Nice work the current version works well.

I don't think it reads 32 bits at once. It's probably more likely that it reads 1 word per cycle like you said.

I like the H_CNT(0) word select for sprite reading so I tried that too. I changed back OBJ range & time to use 1 word same as your previous PPU and these changes below. It works fine too.
````vhdl
OAM : entity work.dpram_dif generic map(8,16,8,16)
port map(
	clock			=> CLK,
	data_a		=> OAM_D,
	address_a	=> OAM_ADDR_WR,
	address_b	=> OAM_ADDR_RD,
	wren_a		=> OAM_WE,
	q_a			=> OAMIO_Q,
	q_b			=> OAM_Q
);
OAM_D <= DI & OAM_latch;
OAM_ADDR_WR <= DBG_OAM_ADDR when ENABLE = '0' else 
				   OAM_ADDR(8 downto 1);
OAM_ADDR_RD <= OAM_ADDR(8 downto 2) & H_CNT(0);
````

````vhdl
if OBJ_RANGE = '1' and FORCE_BLANK = '0' and H_CNT(0) = '1' and DOT_CLKR_CE = '1' then
	OAM_ADDR <= std_logic_vector(unsigned(OAM_ADDR) + 4);
end if;

if OBJ_TIME = '1' and FORCE_BLANK = '0' and DOT_CLKF_CE = '1' then
    OAM_ADDR <= "0" & OAM_TIME_INDEX & "00";
end if;
````

sorgelig, 2019-05-06:
Should i wait for this fix before release? It seems you've made a conclusion.

paulb-nl, 2019-05-10:
Fixed by d280b5affc7e416787bf6af23844aac38ef829d8 Thanks srg320

closed 2019-05-10 by paulb-nl