MiSTer-devel/SNES_MiSTer #72: Graphical glitch on 3D doors in Yoshi's Island

CruelTott, 2019-05-02:
When going through doors the 3D opening animation will glitch.

Here is a video showing the issue: https://www.youtube.com/watch?v=N9Vcla8u8k8

This is towards the end of level 1-4, which I think is the first one with doors.
It's not specific to that level though, I believe it's all doors.

The game version I tested with was: "Super Mario World 2 - Yoshi's Island (USA) (Rev 1)".

James-F2, 2019-05-03:
Completed game SRM attached.
[smw2_100.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/3140789/smw2_100.zip)

James-F2, 2019-05-09:
Here is a cheat zip file for use with the new mister cheat menu, to reach the doors much quicker.
place in: /cheats/snes/
It works for Rev0 and Rev1 of the game.

[Super Mario World 2 - Yoshi's Island (USA).zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/3160439/Super.Mario.World.2.-.Yoshi.s.Island.USA.zip)

James-F2, 2019-05-10:
Fixed.
Thanks srg320!

CruelTott, 2019-05-10:
Thanks for fixing it!

closed 2019-05-10 by CruelTott