MiSTer-devel/SNES_MiSTer #79: CRT horizontal position

James-F2, 2019-05-10:
Position needs a small tweak, the image needs to move left by 1-2 pixel.

@paulb-nl 
Is it somewhere here, like last time?
https://github.com/MiSTer-devel/SNES_MiSTer/blob/8d0f9bf83746b372592ec38c6f713a8c7db2fa00/src/PPU.vhd#L887

paulb-nl, 2019-05-10:
You need to change Hsync just below that to move the image. However I have tested a build with the latest commits and I see no difference to earlier builds. Still only half a pixel difference compared to a SNES.

Could you measure the Hsync length of your SNES console just like you did with the Genesis?

James-F2, 2019-05-10:
Length between hsyncs: 63.5uS
Sync length: 4.625--4.875 uS (narrowest to widest part).

[image]
[image]

paulb-nl, 2019-05-10:
Wow that was fast. Awesome thanks! So Hsync should be 25/26 pixels. Currently it is 25 pixels and that seems to be closest to a real SNES.

James-F2, 2019-05-10:
HSync length may be right, but the screen is not centered.
Can you please try to center it?
iirc, you have a snes too.
Thank you.

paulb-nl, 2019-05-10:
As I have said above, I compared to a SNES and it's perfectly centered. At most half a pixel difference.

Are you comparing with the same input? I am comparing to the same RGB SCART input with SCART switcher.

If you want to change the position just change the H_CNT values here:
https://github.com/MiSTer-devel/SNES_MiSTer/blob/8d0f9bf83746b372592ec38c6f713a8c7db2fa00/src/PPU.vhd#L890
````vhdl
-- 296 is start 
if H_CNT = 296-1 then HSYNC <= '1'; end if;
-- 320 is end
if H_CNT = 320-1 then HSYNC <= '0'; end if;
````

James-F2, 2019-05-10:
Same here, I use scart rgb and a bandridge switcher.
That looks like a one pixel difference.
What values should I set to move the core 1 pixel to the left?

paulb-nl, 2019-05-10:
To move 1 pixel to the left increase H_CNT start and end by 1

You can also try only decreasing start or increasing end by 1 to make Hsync longer.

James-F2, 2019-05-10:
Thank you.
Indeed, that is half a pixel difference.
I'm closing this.

sorgelig, 2019-05-10:
Half pixel difference means the size of hsync is different. So increasing or decreasing the length of HSync by 1 pixel will move the screen by half pixel.

James-F2, 2019-05-10:
Ah, fantastic.
I will try to match the real console by changing the hsync length.

James-F2, 2019-05-10:
https://github.com/MiSTer-devel/SNES_MiSTer/blob/cc2880041f713ea15fb4a47dd4703e082e34a140/src/PPU.vhd#L890

Change to 297 to match a real SNES.

sorgelig, 2019-05-10:
yes, but it's not last release. Will go to next one.

sorgelig, 2019-05-13:
@James-F2
changes in osd.v i've just pushed I think may affect CRT position.
Do you want to add a quick fix for CRT position?  It's related to SNES and Genesis.

James-F2, 2019-05-13:
Yeah I will compile Genesis and test.

James-F2, 2019-05-13:
Genesis on CRT did move at all, it is the same as before.
I will compile and test SNES.

sorgelig, 2019-05-13:
or may be leave it as is. Wait when it will move to the whole pixel and then adjust :)

James-F2, 2019-05-13:
When it will move a whole pixel?

sorgelig, 2019-05-13:
i don't know. Some day, after some tweaks, may be :)

James-F2, 2019-10-14:
Needs to move right by 1 pixel.

closed 2019-10-14 by James-F2