MiSTer-devel/SNES_MiSTer #81: Super Metroid + A Link to the Past Combo Randomizer

mrchrislad, 2019-05-13:
Hi,

I am a huge fan of the Super Metroid / Zelda A Link to the Past Combo, it breathes so much life into both games (if you don't know what I'm talking about, please check https://samus.link )

It creates ExHiRom, and it boots and plays the Super Metroid part great, but you transition to the zelda part of the game by going through a (few different) door in Super Metroid. When you transition it just leads to a black screen.

I have enquired over at the Combo discord, and they suggested it is the transition code not playing nice. 

Is this something you would be able to look into?

MP2E, 2019-05-14:
I'm quite interested in this too. In case it helps any fpga devs, here is the relevant asm snippet for transitions to Zelda

https://github.com/tewtal/alttp_sm_combo_randomizer_rom/blob/master/src/z3/transition.asm

mrchrislad, 2019-05-15:
With the latest core (SNES_20190515) the game no longer boots up - the initial "NINTENDO" logo pops up but looks really funky, and then goes to a black screen. It did work (as my original comment) on the SNES_20190513.
If there is any troubleshooting I can do, please let me know.

sorgelig, 2019-05-15:
I've generated the ROM and it boots OK.
Although i have no idea how to transit to zelda.

MP2E, 2019-05-15:
Thanks for looking into this!

To transition into Zelda, go left from Samus' ship in the beginning. Then through the Blue Door, then in the next room, go left until you reach a wall. Go down from here until you encounter a blue door on you right hand side. Go through this door, and it will transition into Zelda

sorgelig, 2019-05-15:
ok. i see.. black screen

MP2E, 2019-06-10:
Just checking in to state that it still black screens as of 20190609 release. There are no commits to suggest otherwise, but one of the commits fixed Killer Instinct, so I figured it was worth testing.

MP2E, 2019-06-18:
fixed as of 9bc4feaa7aea443fd144baf70f0f1f2ec9957cc8 !

Fantastic! I played it for over an hour with no issues, and switched between games multiple times. Thank you!

mrchrislad, 2019-06-19:
I'm not able to compile the rbf so I can't test myself - I'll have to wait for the release to enjoy, but as it seems to be working for my fellow comboizers, I'll close this issue. Many Thanks to all involved :)

closed 2019-06-19 by mrchrislad