MiSTer-devel/SNES_MiSTer #83: DSP-2 Dungeon Master

Hewhoisred, 2019-05-20:
This game doesn't work properly.  It is the only game that uses the DSP-2 special chip.

It only takes a moment to see the problem.  Shortly after you start the game, entering through the door, you'll quickly see heavy corruption to the graphics, and if you move forward, the game will soon lock up.

attached are some screenshots:
[image]
[image]
[image]

Tested with the USA, EUR and JAP roms.

James-F2, 2019-06-09:
Fixed in 20190609.

Hewhoisred, 2019-06-09:
Confirmed fixed.

closed 2019-06-09 by Hewhoisred