MiSTer-devel/SNES_MiSTer #86: Final Fantasy 2/4 Visual Glitch During Screen Transitions

MordecaiWalfish, 2019-05-26:
There is a small line of messed up pixels near the bottom of the screen throughout the game during screen transitions. You can test this easily by starting a game and going through the intro (~5minutes) and when you are able to control the character, simply walk down through the door below you, and then back up to the room you started in. (example pics attached)
[image]
[image]

I had also noticed that you can see this glitch when exiting battles when the screen transitions back to the normal overhead view.

The following video shows the glitch happening when walking between rooms in the final dungeon, and after a fight: https://youtu.be/4waKEgIZS2o

James-F has confirmed with me that this does not occur on original SNES hardware.

MordecaiWalfish, 2019-05-26:
The glitch happens specifically in the frame between when the screen transition effect ends and your character appears. 
This is the reason the main character is not visible in the screenshots shown, as the glitch goes away the instant the player character appears.

James-F2, 2019-05-28:
I see the same thing on a real 3chip snes.

[image]

7 seconds video:
[3chip snes ff2 transition.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/3228976/3chip.snes.ff2.transition.zip)

MordecaiWalfish, 2019-05-29:
closing this as it seems to be showing up on real snes hardware now that tested with a "3 chip" model

closed 2019-05-29 by MordecaiWalfish