MiSTer-devel/SNES_MiSTer #87: Seth Bling - Super Mario World Jail Break

amoore2600, 2019-05-31:
I have a cart with the [Seth Bling - Super Mario World Jail Break](https://www.youtube.com/watch?v=Ixu8tn__91E) 

This cart and the save ram can be ripped to via a UFO Pro 8 or Analogue NT

From the rip of the cart and save ram, the Jail break can be played on:

- Real SNES hardware with a UFO Pro 8  
- Real SNES with an Everdrive 
- Analogue NT with an UFO Pro 8 
- Analogue NT with an Everdrive
- Analogue NT with Jail Break via an SD card 

However the jail break fails to run on MiSTer.  

I have attached the save ram file.  Matching rom has a check sum crc32 b19ed489

[SUPER MARIOWORLD - JB.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/3239630/SUPER.MARIOWORLD.-.JB.zip)

sorgelig, 2019-05-31:
It's unclear what exactly not working. ROM itself or save?
Note: saves never meant to be transferable. It may or may not be transferred from other platform. It's not a bug.

amoore2600, 2019-05-31:
It's the save that does not work. It nolonger does the jailbreak.

sorgelig, 2019-05-31:
MiSTer doesn't need jailbreaks. It accepts any modified ROM.
There are many SMW mods you can play directly.

paulb-nl, 2019-06-08:
This appears to be an issue with SRAM mirroring.

It works in Bsnes and it mirrors SRAM of Super Mario World to 70-7D:0000-FFFF and F0-FF:0000-FFFF.

Removing "and CA(15) = '0'" from here makes it work: https://github.com/MiSTer-devel/SNES_MiSTer/blob/71bc10007fc690896cf886f10d7b05ce4595b859/src/chip/DSP/DSP_LHRomMap.vhd#L109

sorgelig, 2019-06-09:
should be fixed in today's release

amoore2600, 2019-06-09:
Freaking Awesome!

James-F2, 2019-06-10:
@amoore2600 
Please use the "Close Issue" button.

closed 2019-06-10 by amoore2600