MiSTer-devel/SNES_MiSTer #94: Errors in new translation for Thracia﻿ ﻿776

mabrilG, 2019-06-11:
Have been released a new fan translation for Fire Emblem: Thracia 776. The game star but have corrupted graphics and fonts errors.
This patch game only work in some emulators like is described in the web page of the autors because﻿ the﻿﻿﻿ ROM ﻿has been expanded.
 
Maybe can support this type of roms.

https://forums.serenesforest.net/index.php?/topic/85081-fe5-project-exile-thracia-776-script-and-menu-translation-fully-complete-and-released/

Thanks.

James-F2, 2019-06-11:
> This patch game only work in some emulators

Does it work on real console and sd2snes?
If not, maybe you should improve the patch to work on real console first?

paulb-nl, 2019-06-11:
SD2SNES fix: https://krikzz.com/forum/index.php?topic=7230.0

James-F2, 2019-06-11:
optiroc on discord said it's **reversed ExLoROM**,, no other game does something similar to this.
@sorgelig can update Main_MiSTer to support this kind of rom, or the developer could just convert it to standard ExLoROM for it to work.

sorgelig, 2019-06-11:
why need change the core for a single weird ROM instead to ask the translator make it standard?

Kitrinx, 2019-06-11:
Normally I would agree that adding a one-off exception for a single hack would just be clutter. 

So, after discussing this, it seems the only way make a LoROM game larger than 4mb is to either add a special chip to it, which is difficult and makes repros hard, or to invent a new mapper "ExLoROM". This game being LoROM the authors chose to make it ExLoROM. SNES9X recently added support for this mapping with this kind of scenario in mind, and I think it is likely that we see more of these pop up in the future, now that it is becoming supported by other things.

It's a high profile game and we have support for things like the Star Ocean hack, so maybe it's not that unreasonable to add for now and for the future.

Optiroc, 2019-06-11:
It was my lack of sleep talking when I said “reversed ExLoROM” – this is indeed (the one example of?) ExLoROM. 

I concur with Kitrinx; it is a sound (and simple) extension to regular mode 0x20 that won’t break any existing games, and seems as such a very reasonable addition to SNES_MiSTer.

sorgelig, 2019-06-11:
ok

Kitrinx, 2019-06-11:
Works corrects now, thank you.
[image]

mabrilG, 2019-06-12:
Thank you Sorgelig, now we can play this great game in our dear Mister.

closed 2019-06-12 by mabrilG