MiSTer-devel/SNES_MiSTer #97: Mecarobot Golf - [HDMA related] Flickering during gameplay

MP2E, 2019-06-18:
Here's a clip of it running on the MiSTer, this is the USA ROM. 

https://streamable.com/x8nhg

According to the following article, flickering in this game is often caused by the HDMA timings being off

http://helmet.kafuka.org/accuracy/

James-F2, 2019-06-18:
This is one of the very few games that uses 239p mode, so HDMA might be affect by the non typical resolution.

James-F2, 2019-07-06:
@srg320 
At this stage Mesen-S is more accurate than bsnes and has better debugger than bsnes-plus.
I thought this might give the core the last push with a great reference.
https://github.com/SourMesen/Mesen-S

srg320, 2019-07-06:
Has better debugger  - it's true, but more accurate than bsnes - I'm not sure.

MP2E, 2019-09-25:
I noticed there were some mode 7 changes so I just wanted to report that the game still has the same visual bug as of latest commit

MP2E, 2019-10-06:
This seems to flicker less as of #148 ! However unfortunately, the issue is still present

James-F2, 2019-10-06:
Are we sure it's HDMA related?
Seems to be Mode7 not calculated fast enough since I see a flicker of the entire map from above for a frame... or so it appears to me.

MP2E, 2019-10-06:
I'm not sure if it's hdma related at this point. It's hard to say

James-F2, 2019-10-07:
If a background layer (Mode 7) is inserted from the middle of the screen, it is some kind of DMA/HDMA related trick.

srg320, 2019-10-07:
@James-F2
This is it..

paulb-nl, 2019-10-20:
It seems the flicker is caused by an IRQ/DRAM refresh issue. There is an IRQ in the mode 7 layer that is close to the refresh and sometimes this IRQ is delayed until after the refresh or the IRQ is not triggered at all. That does not happen in Mesen-S.

The flicker does not happen on a build with DRAM refresh disabled.

Here are videos which show yellow lines where 'IsIRQInterrupt' is high.
Refresh enabled: [https://youtu.be/faGqxFDCIlw](https://youtu.be/faGqxFDCIlw)

Refresh disabled: [https://youtu.be/mYBNUaqAJeU](https://youtu.be/mYBNUaqAJeU)

You can pause the video and press , and . to advance frames.

James-F2, 2020-01-05:
https://github.com/MiSTer-devel/SNES_MiSTer/pull/177
This seems to fix that. :)

srg320, 2020-01-05:
Oh, unexpectedly.

James-F2, 2020-01-05:
Could be just my build.
Needs confirmation by another person.

MP2E, 2020-01-05:
I'm amazed! It does seem to be fixed by that commit. Completely eliminated all flickering. Thanks again!

closed 2020-01-05 by MP2E