MiSTer-devel/SNES_MiSTer #100: Combat Tribes - flicker/transparency that happens when 2 characters over lap

amoore2600, 2019-06-19:
There is a strange flicker/transparency that happens when 2 characters over lap. I have included a video that show and example of what happens on MiSTer vs SuperNT.

https://youtu.be/rWvjD3atY3g

James-F2, 2019-06-19:
Happens even in title screens.

https://youtu.be/Tcfm7iDEeu0

MP2E, 2019-06-19:
I can confirm this issue on latest commit. I tested this on a 1chip-03 SNES with an sd2snes to make sure that the flicker does not happen there, and it does not.

James-F2, 2019-06-19:
A little guess here,
Seems like an issue with sprite priority with combination of improper game code.

paulb-nl, 2019-06-19:
The issue appears once when there are too many sprites on one line. Currently the core does not reset the sprite overflow flags at the end of Vblank if Force-blank is enabled. Probably because Fullsnes Nocash doc says this: "The overflow flags are cleared at end of V-Blank, but NOT during forced blank!"

Higan and Mesen-S seem to reset the flags regardless of Force-blank.

The game enables Force-blank through Vblank until scanline 13 so the overflow flags are never reset.

Removing "and FORCE_BLANK = '0'" here fixes the issue: https://github.com/MiSTer-devel/SNES_MiSTer/blob/067dcc90d236c4f23db65c56530d3ad9866853a8/src/PPU.vhd#L1485-L1487

James-F2, 2019-06-19:
Why would Higan and Mesen-S choose to do something the real console does not?

paulb-nl, 2019-06-19:
The real console most likely also resets the flags regardless of Force-blank otherwise you would see this issue on a real console. There would have to be some test rom made to be sure.

James-F2, 2019-06-19:
I meant to ask is, did Higan and Mesen-S developers misunderstand this part of the Fullsnes Nocash documentation?
I can see the same flickering in Higan too but not on the real console.

sorgelig, 2019-06-19:
Probably doc is correct. It's just means that flag is cleared during V-Blank regardless force blank flag. 
Doc wanted to tell that force blank doesn't clear the flag.

paulb-nl, 2019-06-19:
@James-F2 You mean only flickering in-game right? Do you really not see any flickering during the in-game demo on real console? There are a lot of sprites there with the 2 players and enemies. Especially when they are lying on the floor.

James-F2, 2019-06-19:
I have compiled without ""and FORCE_BLANK = '0' then "" as you suggested.
There is some flickering when there are a lot of enemies on the floor (demo mode) but it matches the real console now.

MP2E, 2019-06-20:
Tried the latest commit, it is fixed here too. Looks just like my 1chip SNES

James-F2, 2019-06-21:
@amoore2600 
Please use the "close issue" button.

closed 2019-06-21 by amoore2600