MiSTer-devel/SNES_MiSTer #102: Tengai Makyou Zero (SPC7110 IC support) possible?

bootsector, 2019-06-21:
It would be great if SNES core could support this game, that uses the SPC7110 RTC chip.

This is one of the greatest RPG games on the Super Famicom. It features an awesome english translation: https://www.romhacking.net/translations/3243/

🙏

MP2E, 2019-09-18:
Looks like this is implemented as of d806c50aa9340a35f0a73c7dc38bef72122a3189

bootsector, 2019-09-18:
Tested with latest RBFs posted at Classic Gaming's mister-general channel. I've only got a black screen after loading the translated ROM. Any clues?

MP2E, 2019-09-22:
Try the latest release, should run perfectly. It is for me, at least

bootsector, 2019-09-22:
@srg320 I only have three words for ya: I LOVE YOU!!!! <3
Thank you, thank you, thank you!!!! 🙏 
It even worked with my old save!!!!
I'm happily closing this!
THANK YOU!!!!!

closed 2019-09-22 by bootsector