MiSTer-devel/SNES_MiSTer #104: Donkey Kong Country 3 - Random graphical corruption in intro and title screen

MP2E, 2019-06-22:
This one is tricky because it is random. Sometimes the intro and title screen play with no graphical corruption. But about half the time or a bit more, there are random graphical artifacts in the title screen and in the part where the characters are turning the wheel for the music box.

https://gfycat.com/ambitiousgoodnaturedadeliepenguin

In this clip, the music box part plays with no glitches, but minor glitches can be seen in the title screen.

At first I thought this was a ROM file corruption issue, but the checksum matches No-Intro and SmokeMonster's packs. I also tried the exact same rom on an sd2snes with a 1chip-03 SNES and could not replicate the issue after many resets. I also have a real Donkey Kong Country 3 cart that does not show the issue

MP2E, 2019-06-23:
Just looked into this a bit more and noticed that every tile technically has the right graphics from what I can tell, it is simply the palette that is wrong. I have updated the title of this issue to reflect this. Which tiles get miscolored does seem to be random, though.

James-F2, 2019-06-23:
I've been trying for an hour now, I can't reproduce this glitch.

Can this be a bad synthezation on quartus side?
Maybe sdram related?

Try this build:
[SNES_Cybernator_20190622.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/3317465/SNES_Cybernator_20190622.zip)

MP2E, 2019-06-23:
I tried your build and looped through the intro and title screen over 20 times without any issues. Before it would happen very, very often, so I think it is safe to say I made a bad build somehow. Thank you for looking into this!

I recently tested my SDRAM for 40 minutes using the auto test in the memtest utility and it found 0 errors, so I think it is just the build I created... But I only used the latest sources. So maybe I need to change my Quartus version.

closed 2019-06-23 by MP2E

MP2E, 2019-06-23:
Aha. It turns out I didn't clean the project folder before building. I will be more careful in the future.