MiSTer-devel/SNES_MiSTer #105: Currently failing DMA/HDMA test roms

MP2E, 2019-06-26:
byuu, creator of bsnes/higan, has a bunch of tests he made while creating the emulator publicly available with source code at the following page:

https://snescentral.com/homebrew.php

I ended up downloading a bunch of the HDMA/DMA tests, and ran them. Red screen is failure, blue screen is pass. Here are the results

test_dma - Failure
test_dmasync - Pass
test_dmavalid - Failure
test_hdma - Failure
test_hdmasync - Failure
test_hdmatiming - Failure
test_mdrhdma - Pass

These tests can be found on this particular page: https://snescentral.com/article.php?id=1123
I used this link https://snescentral.com/1/1/2/1123/test_mdrhdma.zip

I also found a forum post where he describes some of the expected behaviors that are undocumented, specifically related to these tests

https://forums.nesdev.com/viewtopic.php?p=236984#p236984

MP2E, 2019-06-26:
Just tested these on a 1chip-03 SNES: every tests passes with the exception of test_hdmatiming.

test_hdmasync also seemed to fail once on my 1chip-03, but on a second try, it passed.

These tests should be run on a non-1chip SNES for verification

srg320, 2019-06-26:
I achieve this for about year, but unfortunately so far unsuccessfully.

James-F2, 2019-06-26:
### Snes Jr (1chip):
dma (5 out of 5 tests, pass)
dmasync (5/5)
dmavalid (5/5)
hdma (18/20 pass)
hdmasync (2/10)
hdmatiming (4/18)
mdrhdma (10/10)

**GPM-01 (3chip)**
dma (10/10)
dmasync (10/10)
dmavalid (10/10)
hdma (18/20)
hdmasync (4/10)
hdmatiming (4/20)
mdrhdma (10/10)

MP2E, 2019-06-26:
After some discussion and testing across a few different consoles, I do not think the HDMA tests are worth pursuing seriously because they fail intermittently on real hardware. Even the Super NT fails all the HDMA tests except for sometimes passing the hdmasync one.

The DMA tests however, do look good

MP2E, 2019-06-27:
> I achieve this for about year, but unfortunately so far unsuccessfully.

Ah, did not know that you were aware. I think the games are most important, and it already plays everything so well! Thank you!

I consider this a low priority, I brought it up in the hopes that these tests would help for debugging more than games with smaller glitches.

James-F2, 2019-07-01:
**test_dma** now passes after the Bugs Bunny fix.

MP2E, 2019-07-01:
Confirmed! Since there is only 1 valid test left to pass, I have changed the title of this issue to reflect that

MP2E, 2019-07-03:
Fixed! amazing work as usual :)

James-F2, 2019-09-28:
@MP2E 
I'd like to reopen this issue as none of the HDMA tests pass even once.
Maybe only the 'hdma' test should pass like the real console or bsnes, but 'hdmasync' and 'hdmatiming' are problematic on the real console too.

MP2E, 2019-09-28:
Reopened !

MP2E, 2019-10-06:
Now the HDMA tests seem to pass as of #148

closed 2019-10-06 by MP2E

James-F2, 2019-10-06:
Yep, all HDMA tests pass,, great work @srg320 !