MiSTer-devel/SNES_MiSTer #107: Marvelous - Another Treasure Island - Text renders incorrectly (hicolor math)

MP2E, 2019-06-27:
As the title states, in Marvelous - Another Treasure Island, it seems that some of the text ingame renders incorrectly, and sometimes the logo of the game that appears between text boxes has a few pixels that are colored incorrectly. This is likely the same issue, it is picky about the hires color add/sub effects

https://imgur.com/a/6Ssw4i9

I used an english translation to test the game which can be found here, but I also tested it on my SNES 1chip and didn't see the issue

https://www.romhacking.net/translations/2558/

James-F2, 2019-06-27:
Can't reproduce.
The image above looks like the Super NT bug, but I can't reproduce on MiSTer.

James-F2, 2019-06-27:
Another problem is this game reset to black screen.

paulb-nl, 2019-06-27:
Found the cause. The text displays properly if the previous MAIN_COLOR is used for SUB_* color math instead of current MAIN_COLOR

https://github.com/MiSTer-devel/SNES_MiSTer/blob/58c42dd7b004cf313231131e3ace407b21baed21/src/PPU.vhd#L2244-L2247

MP2E, 2019-06-30:
PR #109 fully fixes the text and logo issues across the versions of Marvelous, however, as @James-F2 notes, if you reset the game gets stuck at a black screen. Changing the title of the issue to reflect the remaining problem

James-F2, 2019-06-30:
Better close this issue and open a new one for the reset to black.

MP2E, 2019-06-30:
will do!

closed 2019-06-30 by MP2E