MiSTer-devel/SNES_MiSTer #113: Magical Drop - Endless Mode Hangs at Game Over screen

MP2E, 2019-07-04:
In Magical Drop, whether you use the original Super Famicom ROM, or the translated english version, if you play Endless Mode, when you eventually lose the game will be locked and not allow you to continue. Here is a clip:

https://streamable.com/e5cll

paulb-nl, 2019-07-04:
Some relevant information:

> This WIP rewrites huge swaths of the SNES
> DSP core to separate internal registers from underlying RAM. This allows me to
> emulate the randomized memory registers at power-on, while not breaking games
> due to also randomizing the actual DSP registers. This is part of Jonas Quinn's
> findings about a year and a half ago, which is aimed at improving the chances
> Magical Drop doesn't crash, as on real hardware

https://gitlab.com/higan/higan/commit/8ed164c605efa7b4f48aad64fb28d745a793b492

https://helmet.kafuka.org/bboard/thread.php?id=108
https://helmet.kafuka.org/byuubackup2/viewtopic.php@f=4&t=1668&start=10.html

MP2E, 2019-09-22:
Addressed by #137 ! Thank you srg320!

closed 2019-09-22 by MP2E