MiSTer-devel/SNES_MiSTer #114: The SNES core no longer works from a cold start.

tonurics, 2019-07-04:
MiSTer v190627
OS v190627

I've been having trouble with the last two releases of the SNES core. Here's my experiences with them:

SNES_20190609.rbf
Works correctly every time, no issues. [I downgraded to it after not being able to use SNES_20190627.rbf]

SNES_20190627.rbf
From a cold start nothing loads [blank screen, always]. Starting another core [e.g. Genesis] first, then switching to the SNES core will sometimes cause it to load for a few seconds, before showing visible/audio corruption before crashing [blank screen, random]. [Never been able to get to a playable state.]

SNES_20190627.rbf
From a cold start nothing loads [blank screen, always]. If I start the NES or Genesis core first, then the SNES core appears to work normally [repeatable]. If I start the Gameboy core first, then the SNES core I immediately get visible corruption and an eventual freeze [repeatable]. Switching from the SNES core, to another code, and then back to the SNES core sometimes crashes [blank screen, random].

sorgelig, 2019-07-04:
I don't see such problem.
But core is large and can be unstable if you don't use the fan.

tonurics, 2019-07-04:
Sorry, just noticed a typo: the second "SNES_20190627.rbf" listed should be "SNES_20190703.rbf", it's the version that works after loading the NES or Genesis core.

I'm using a Noctua fan, along with a heatsink. Since loading other cores before the SNES core gets it working: my uneducated guess would be that something changed with how things are loaded into memory starting with SNES_20190627.rbf.

Is there anything I can do to help diagnose? [I have a Infrared Thermometer, and I can use it to get a reading, but I feel pretty safe anecdotally saying that it's not running hot.] Or do you have any other ideas?

MP2E, 2019-07-04:
Please try this build to see if it fixes the issue:

https://drive.google.com/file/d/1YibUgT3ymwyBVVj3PNV_c1bOS5vY2Nca/view?usp=sharing

This is the latest changes, the only difference is that this build was built with a different fitter seed, which can help sometimes

tonurics, 2019-07-04:
Thanks for the alternate build.

[Being extra verbose to help figure out what might be going on.]

Turned on the MiSTer, looked up IP Address from the menu, copied SNES_DMAVALID_20190703.rbf to SDcard via network/SFTP. Started SNES_DMAVALID_20190703.rbf: works. 

Power cycled MiSTer, started SNES_DMAVALID_20190703.rbf: blank screen. Repeated a couple of times: no change. It's now behaving the same as the normal SNES_20190703.rbf release.

Interestingly, if I just switch back and forth between SNES_DMAVALID_20190703.rbf and SNES_20190703.rbf both cores will start working. Sometimes it works immediately; but not always, but eventually after a few switches back and forth they will both start working. [Starting the NES or Genesis core first seems to always work.]

The core appears to continue to work after soft reboots, but not "cold" reboots. Also leaving the MiSTer run idle for a little while before trying to start the core, appear to have no effect. Not sure what magic ingredient got it working the first time around...

James-F2, 2019-07-04:
Timing is becoming very tight with this core.
Some fitter seed numbers will produce a working compilation, yet others will not.

sorgelig, 2019-07-04:
You may try latest changes - may be it solves the issue

tonurics, 2019-07-05:
Thanks for the replies. I look forward to testing the changes out.

I looked briefly at trying to assemble my own *.rbf files from the source here. I came across Quartus Prime [assuming that's what is need] but discovered some dependency issues with it on my Linux workstation. So I figured I would wait for one of you kind gentlemen; who already have working build environments and actually know what you are doing, to eventually provide one. ;)

MP2E, 2019-07-05:
https://drive.google.com/open?id=1F9FLvfSyIOlWH-vlXN0eW_34MYHwY_aQ

Here's the latest build, I've tested it across a bunch of games and it's stable here

sorgelig, 2019-07-05:
> Here's the latest build, I've tested it across a bunch of games and it's stable here

did you compile with default SEED 1?

MP2E, 2019-07-05:
No I used 4271993 again but I can try seed 1. I didn't try seed 1 with the newest changes, but I will now to check

MP2E, 2019-07-05:
Yup Seed 1 works great, no timing issues at all on latest commit for me

James-F2, 2019-07-05:
Compiled with SEED 1, everything is okay.

tonurics, 2019-07-05:
Huzzah! No issues with the new release on my end. It's loading every time now: power cycles, "cold" reboots and starting after the Gameboy core.

My thanks to everyone on this issue. With an extra shout out to MP2E for providing dev releases, so I could help test.

simbin, 2019-07-24:
I'm experiencing same issues tonurics was, but with very latest files today. I've tested a few other builds, including older July 4th build MP2E posted here, but no luck.

tonurics, 2019-08-08:
Sorry, haven't had a chance to follow up until now.

But release 20190719 is working without issue for me; reboots and power cycles are all good.

closed 2019-08-08 by sorgelig