MiSTer-devel/SNES_MiSTer #116: Chrono Trigger - Demo loop sometimes exits early even with no input

MP2E, 2019-07-05:
If you leave Chrono Trigger running, it will play a bunch of demo sequences after the title screen animation. At random, but almost always in the 2nd or 3rd loop of the demo sequence, the demo will be abruptly ended and forced back to title screen without any input from the user, as if the user had pressed a button. This behavior was tested on console and did not appear after 8+ loops.

Multiple users confirmed with different builds, to make sure it was not a one off Quartus timing issue.

James-F2, 2019-07-05:
What could possibly stop the demo besides a controller input?

MP2E, 2019-09-18:
I took another look at this to see if I could narrow it down, but unfortunately the bug remains mysterious. Even unplugging the USB hub entirely the bug happens in the newest builds in about the same place as before.

Also, terminator2k2 reported that he had Chrono Trigger crash at this screen in the demo loop
https://cdn.discordapp.com/attachments/497145588446396416/623810940093202432/MiSTer_FPGA_GunSmoke_-_By_Jotego_Screenshot_2019-09-18_10-21-27.png

MP2E, 2019-09-22:
Interestingly, one of the timing changes between either srg320's new PR at #137  or the commits made in the past 2 days by Sorgelig seemingly made this happen *much* less often. We thought it was fixed, but unfortunately I managed to trigger it after like 8 loops

This could imply that timing issues were causing it, but I am not certain

perhaps this commit helped? 8022fb7e2ba7073309df7d37c97843c98e40c8c6

Kitrinx, 2019-09-24:
I had a suspicion that this was actually being caused by input. I put it on signal tap, and sure enough, the moment that the intro reset it's loop, signaltap triggered and showed button presses on both controller 1 and 2 happening erroneously.

[image]

MP2E, 2019-09-26:
Addressed by 78aada129a622d39d1443bc64ddf7421810f39cc ! Thanks again :)

closed 2019-09-26 by MP2E