MiSTer-devel/SNES_MiSTer #119: Super Mario All-Stars + Super Mario World - Game glitches after going down a pipe in Super Mario World

KawaiiDinosaur, 2019-07-14:
[https://imgur.com/a/9p3lQYy](url)

Name: Super Mario All-Stars + Super Mario World (USA).sfc
Size:	2 MB (2,621,440 bytes)
SHA-1: d245e41a2b590f7d63666b0772cbddfb26f254a2
MD5: 7ecaf3e2e021c8a836041c85886f4594
CRC32: f84305b1

James-F2, 2019-07-14:
The latest official version is broken.
Please try this version and report.
[SNES_LDA.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/3390119/SNES_LDA.zip)

KawaiiDinosaur, 2019-07-14:
> 
> 
> Always mention what core version are you using, and test previous version too.
> The latest official version is broken.
> Please try this version and report.
> [SNES_LDA.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/3390119/SNES_LDA.zip)

Core verion is SNES_20190703. Will try this and report back.

KawaiiDinosaur, 2019-07-14:
It's working flawlessly on the LDA building. Thanks for the heads up.

closed 2019-07-14 by KawaiiDinosaur