MiSTer-devel/SNES_MiSTer #122: Super Castlevania IV - Demo loop hangs randomly

nocmonkey, 2019-07-22:
Demo loop plays through a random number of times and then hangs at different parts of the demo.  It doesn't hang on SNES_20190609.rbf
Hangs on SNES_20190719.rbf
Those are the only releases I tested Super Castlevania IV on.
rom md5sum: 094f035993e9724647b61ddcba1e9a7a

Let me know if you need more info or if this is the wrong place to post.  Thanks!

MP2E, 2019-07-22:
Hmm, just let my MiSTer loop for about 4 loops and didn't see anything off. Do you mean it freezes entirely when you say hang? or something else?

nocmonkey, 2019-07-22:
It freezes at a frame at different points.  It's very random, it could happen in 30 minutes or even an hour.  The MiSTer menu is still responsive so I can reset or reboot when the game video freezes.  I leave my MiSTer on 24/7 and just have it play a demo loop when I'm not using it.  Super Metroid is usually what I have loop but recently switched it up with Super Castlevania IV.

James-F2, 2019-07-22:
Running for 2 hours now, didn't freeze yet.

nocmonkey, 2019-07-22:
Yeah, mine finally froze just now after 2 hours and 20 minutes. haha  Kinda hard to work with that.  If there is anything I can do over ssh or something to get any data for you?  I'll leave the MiSTer at the freezed screen.  Other than that, sorry for the bug report, I didn't think it would take that long to glitch. haha

James-F2, 2019-07-22:
Okay, I'm running Super Castlevania IV intro from 9:30 till now 19:30,,, no freeze.
I think the problem is on your side @nocmonkey , maybe the sdram module?

nocmonkey, 2019-07-22:
I was thinking the same thing last night.  I'll order a replacement sdram module and give it a shot.  Thanks!

closed 2019-07-22 by nocmonkey