MiSTer-devel/SNES_MiSTer #125: Colors are wrong for "T2 - The Arcade Game"

amoore2600, 2019-08-22:
The colors are wrong for "T2 - The Arcade Game"

The colors are: 

[image]

The colors should be:
[image]

MP2E, 2019-08-22:
Strange, I can't seem to reproduce this bug. It looks like the bottom image to me, I played for a few minutes. Does this occur later into the game?

amoore2600, 2019-08-22:
**Additional information:**

ROM checksum:
/media/fat/SNES/6 Personal# md5sum T2\ -\ The\ Arcade\ Game\ \(USA\).SFC 
49a4cf748b491f6e7bdbfc86f2cf4d16  T2 - The Arcade Game (USA).SFC

Core version: SNES_20190719.rbf

Wrong colors happens from the beginning of the game.

MP2E, 2019-09-02:
Still can't seem to reproduce on my end, I've tried like 5 different versions. Not really sure what could cause this kind of variance..

amoore2600, 2019-09-02:
Want me to send you my ROM?

MP2E, 2019-09-04:
I've got the same one according to md5sum

49a4cf748b491f6e7bdbfc86f2cf4d16  T2 - The Arcade Game (USA).sfc

I'll ask in the classic gaming discord and see if others see the same thing

amoore2600, 2020-02-08:
Its taken me a while to figure out how to reproduce this bug but its definitely still here. I was wrong in thinking it was an HDMI cable issue that was originally causing it. 

The bug can be reproduced by:

1. Setting "SNAC" on the OSD for Serial  
2. Save settings
3. Reboot or Reset
4 Start the game using the SNAC controller 

Here is a video of the bug: https://youtu.be/bBgXDWc5sAY

paulb-nl, 2020-02-08:
On your video you started the game with a Super Scope and I could reproduce that.  It does not happen with a normal SNES controller with SNAC.

Guess what happens when you start the game with the Scope on a real SNES.

[image]

amoore2600, 2020-02-08:
WOW!!!! I totally did not expect that. Closing!!!

closed 2020-02-08 by amoore2600