MiSTer-devel/SNES_MiSTer #129: Glitches/lockup with Contra III: The Alien Wars

steven-a-wilson, 2019-08-25:
Contra III: The Alien Wars locks up while playing the first level with releases 20190823 and 20190824. It doesn't take a fixed amount of time or occur at a specific location within the level, so I'm not sure what the actual conditions are. But it often happens within a few seconds. There is often graphics corruption associated with this, and the screen sometimes goes black when the game locks up. In one case the screen went black and then the game continued as though I had finished the first level, then showed graphical corruption and locked up after starting the second level.

Similar glitches with other games have also been reported in the forum recently. Maybe these are the same bug, maybe not.

Release 20190719 seems to run the game normally.
Release 20190820 just shows a black screen after loading the game.

I tried to bisect this, but I found that the issue doesn't occur with cores built on my own machine (with Quartus Prime 18.1.0 Lite Edition, after symlinking sys/pll_q17.qip to sys/pll_q18.qip to get around a build failure).

steven-a-wilson, 2019-08-26:
After arranging for my board to be cooler, I was able to finish the first level of the game in all three of the "bad" builds (20190820, 20190823, 20190824). So I guess this was somehow a perfect storm of timing failure and not exactly a code defect. I'm not familiar enough with how this sort of thing is specified/simulated/checked to know if this is still considered a bug or not.

steven-a-wilson, 2019-08-29:
There are two variables that independently "fixed" it, though:

1) Changing to a different core or different build of the SNES core (even with the same source)

2) Lowering the temperature

I was thinking that there might be something like a missing or incorrect timing constraint, causing the fitter to accept layouts that actually have incorrect timing at an in-spec temperature. But I guess without a way to accurately measure the temperature (and my understanding is that Intel/Altera reserves on-die temperature sensors for its more expensive device families), it's hard to say whether it was in-spec.

Anyway, I'm fine with closing this if nobody wants to investigate further.

sorgelig, 2019-08-29:
> I was thinking that there might be something like a missing or incorrect timing constraint, causing the fitter to accept layouts that actually have incorrect timing at an in-spec temperature. But I [...]

you are welcome to submit the patch with more constraints.

steven-a-wilson, 2019-08-30:
I'm closing this because after experimenting with it a bit more I consider it more likely than not that temperature was out of spec when I saw this. I don't have a rigorous methodology or anything, but ambient is currently pretty warm, I'm pretty sure the heatsink is hot enough to cause actual burns on my fingers in a matter of seconds, and I've been playing Contra III on the 20190820 build for 20+ minutes with no signs of instability.

closed 2019-08-30 by steven-a-wilson