MiSTer-devel/SNES_MiSTer #131: DMA RAM access

gyurco, 2019-09-10:
Not a real issue, but a question about DMA transfers:
Seems when the core does DMA from WSRAM to VRAM, CPURD_N and PAWR_N are asserted at the same phase. This means that the read cycle of the source WSRAM and the write cycle of the destination VRAM are the same. Is it really happens on the real hardware? Doing a read+write in 4 master clock cycles is unbelievable for me. Probably PAWR_N should be asserted in the other clock phase?

srg320, 2019-09-10:
I don’t know what is really happening in real hardware. I used the logical analyzer traces from this [forum](https://forums.nesdev.com/viewtopic.php?f=12&t=13088). For example, in LISTING8.txt from line 1477 starts DMA transfer and CPURD and PAWR are asserted simultaneously. 
If someone will share more detailed information about the work of DMA I would be very grateful.

gyurco, 2019-09-10:
Indeed. Would be interesting to see the PPU's VRAM write enable lines, too. Two RAM cycles in an 21MHz/4 period seems too fast for me. Then they could use a much faster CPU.

gyurco, 2019-09-10:
Then I choose to drive VRAM's write enables after PAWR_N asserted:
https://github.com/gyurco/SNES_MiSTer/commit/ec6709262bc4aff8590c7db5425cbe39fe851f54
Gave it limited testing, seems to work. Of course it's not needed for the BRAM-based VRAM, but I wonder how could it work on your original dev board if you used the external SRAM chips.

sorgelig, 2019-09-11:
@gyurco did you get the MiSTer? ;)

srg320, 2019-09-11:
SYSCLK duration is 3/3 (fast rom and B bus), 3/5(slow rom/ram), 3/9(joystick ports) and 4/4 (DMA) cycles MCLK 21MHz. CPUWR/RD and PAWR/RD signals is 3, 5, 9 and 4 cycles respectively.
So, for DMA 21MHz/4=~200ns is normal for SRAM  chips of those years. Regarding PPU, I think the data is first written to the buffer, and then to the memory , by analogy with reading. VRAM access by PPU rendering is 1 dot clock = 4 cycles MCLK.

gyurco, 2019-09-11:
@sorgelig no, this actually works on MiST :)

@srg320 yes, you're right. It's 200ns, which is enough for SRAMs and probably adequate for cartridge ROMs of that time, but for only one cycle. With faster SRAMs it could even work with both read and write, however I doubt it would work with mask ROMs. The patch I linked exactly does this presumed buffering: it registers the input from DI, and asserts VRAMx_WRITEs after that 200ns period.

srg320, 2019-09-11:
Adrress bus and /CS are asserted at the beginning of the SYSCLK, CPURD connected to rom /OE and asserted later. May be the internal logic of the rom chip have time to select the desired cell and RD signal only open data bus to output. But I could be wrong.

gyurco, 2019-09-11:
Probably this is the solution. Too bad I couldn't find a datasheet for S-WRAM to see the data output valid time from /OE asserted.

closed 2019-09-12 by gyurco