MiSTer-devel/SNES_MiSTer #136: SFC player support

vinz2k, 2019-09-20:
Hello,

I think the NES core has support for NSF Player (but i’ve never test). Should it be possible to have a SFC Player for the SNES core ? (Maybe inspired with the SFC Player from Analogue SuperNT).

Thanks for all your work,

Snes core for MiSTer is for me a candy everyday 🥰.

Regards

srg320, 2019-09-21:
It’s possible and I thought about it. 
But this will not happen until the core compiles stably. We are working to resolve this issue.

sorgelig, 2019-09-21:
Linux on MiSTer already supports VGM/VGZ playback. 
You can download favorite tunes and play it on MiSTer in Linux using Midnight Commander.

vinz2k, 2019-09-21:
Thanks for the info :) 

I’ll try it

vinz2k, 2020-01-06:
Hello,

Any news regarding this topic ?

Or should we close the issue ?

Regards

Beatatester, 2020-10-05:
If MiSTer does get an SPC player, it would be nice if it could play SPC2 files, since those are finally starting to catch on. The format seems to be [better suited for FPGA and real hardware](http://blog.kevtris.org/blogfiles/spc2_file_specification_v1.txt) than SPC in some ways.

Chocobobobo, 2020-10-18:
Your wish has been granted, even if there is no interface, and some weird scratch when you load a new track, and doesn't use sp2 files.

vinz2k, 2020-10-19:
Oh really ?

So we can load SPC file directly from the SNES core and they will play ?
Is that right ?

vinz2k, 2020-10-27:
- Simple option to play SPC files. Audio only, without interface (srg320)

James-F2, 2020-11-10:
Can this be closed now?

vinz2k, 2020-11-10:
Sure

closed 2020-11-10 by sorgelig