MiSTer-devel/SNES_MiSTer #138: Far East of Eden Zero - Tengai Makyou Zero Problems

Hewhoisred, 2019-09-22:
On the attract mode (intro) for the game:

Optiroc said:
> The first thing I noticed is that during the intro sequence, the battle screen with a large samurai and maneki-neko cat a lot of things are wrong:
> 
> - The hand waves erratically
> - The tail is absent (should be a string of jointed balls)
> - The "Blam!" text zooms erratically and seems to show a random graphic (sometimes "Blam!", but also "Mew!" and some other things)

You'll notice the tail attack is using the paw sprite

Here is some (somewhat poor) footage of how it's supposed to look:
[Link](https://youtu.be/siBUk4CYeHY?t=30)

There is also some problem with what looks like garbage data in the battle sprites:
[image]

Hewhoisred, 2019-09-22:
Some of the geometric battle attacks look off as well.  This is from the intro too:
[image]

Hewhoisred, 2019-09-22:
just a small update:  I can't seem to reproduce the garbage sprite problem.  On a fresh power up of MiSTer that seems to not be a problem.  Maybe something was lingering in memory to cause the problem.

M-Walrus, 2019-09-23:
I've recorded some examples of how this looks on the last official MiSTer SNES build ( 2019 09 21 ):

https://youtu.be/C3OzzGDNb1c?t=59

https://youtu.be/C3OzzGDNb1c?t=309

https://youtu.be/C3OzzGDNb1c?t=558

clearly shows the cat's ball-and-chain tail is missing, and the animation of the paw is very jerky.
text at the bottom looks like it's supposed to be random each time, but may have some scaling issues.

Optiroc, 2019-09-23:
Indeed, the graphic shown in the "Samurai vs Cat" screen during the intro is randomized, but the other issues stand. It seems to be a general "math issue", so my best guess is that it relates to the SPC7110 on board ALU. 

Since the initial power-on test passes, these are all cases not excercised by that test, or the results are not ready when the games read them (which would imply that the power-on test but not the game engine honors the ALU BUSY registers before reading out results).

I noticed (probably) related issues in a few other places during my quick test:
- During the in-game intro at the gate (after registering time and birth date), when the antagonist throws his sword the movement/transformations are glitchy.
- On the "almanac" screen (press X in game, then select "Almanac") the biorythm curve is drawn wrong; all pixels beneath the baseline is drawn way down to the bottom of the screen.

srg320, 2019-09-23:
If issues are detected during the game, please attach a save file and, if possible, a photo or video of this place. Thanks.

Optiroc, 2019-09-23:
@srg320 I’ll fill in some more details during the day. I’ll try to confirm if the game reads out ALU results without checking the BUSY flag today. If so, the MiSTer core will have to have results ready as quick (or quicker) than original hardware. Is it possible that the current implementation takes more cycles before ALU results are done than real SPC7110?

srg320, 2019-09-23:
@Optiroc, it is possible, if you know the specific number of cycles. But I think problem in signed mul/div, self test program only checks unsigned.

Optiroc, 2019-09-23:
@srg320 Ah, that would make a lot of sense, considering how this graph is drawn (accessible anywhere in the game when you control the player, press X then select Almanac from the menu).

bsnes:
[image]

MiSTer_SNES_20190921:
[image]

How the negative values on the graph are drawn is a good indication that something is off with signed operations. ;)

Here is a save file that simply puts you in the beginning of the game, where the X->Almanac screen can be accessed.
[tmz_english_beginning.sav.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/3641168/tmz_english_beginning.sav.zip)

MP2E, 2019-09-25:
I believe all of the issues detailed in this thread have been fixed, if someone could confirm I think this issue can be closed

Hewhoisred, 2019-09-25:
Looks good.  Thanks srg.  Closing.

closed 2019-09-25 by Hewhoisred