MiSTer-devel/SNES_MiSTer #139: SPC7110 RTC behavior

Optiroc, 2019-09-23:
As of MiSTer_SNES_20190921 the SPC7110 will always report the MiSTer's real time when reading the RTC time. (Except if you have set the time with RTC writes during the current session, which the game lets you do when creating a save. In that case, it seems the time you've set is used until reboot.)

While using the "real world" time is possibly a sane implementation of the SPC7110 RTC, it breaks the possibilty to set the game time arbitrarily when starting a new game. This is important, because in Tengai Makyou Zero certain events only happen during 1996 (or 2018 in the fan translation), so you want to experiment with having the game play out during those years.

Emulators commonly implement the SPC7110 RTC by storing the difference between real time and the time you set when writing the RTC time. This difference is appended to the battery save data, and when reading the SPC7110 RTC the diff is added to the operating system's real/current time.

sorgelig, 2019-09-23:
MiSTer is not emulator. It has no GUI where you can easily set the time. 
I only see one possible solution: disconnect MiSTer from internet (to prevent update), and then in Linux screen set the desired time. 
Then you can start the SNES core and play the game.

Optiroc, 2019-09-23:
@sorgelig The interface for setting the SPC7110 RTC (ie the virtual/emulated one) is accessed in game, and performs writes to the RTC to set the time, so no extra interface is needed. 

You don’t want this function to set the real MiSTer time, so storing the offset is rather an elegant and simple solution.

closed 2020-01-22 by sorgelig