MiSTer-devel/SNES_MiSTer #146: Bad Apple Demo, will start with no audio randomly.

James-F2, 2019-09-28:
https://www51.zippyshare.com/v/ttnvGs7Z/file.html
Sometimes the audio will not play at random when loading the rom.

srg320, 2019-09-28:
Without this commit 0aafeb3568eb13281bca628a10946b2c6fc46c9b demo has a sound always, otherwise Magical Drop freezes in Endless Mode. I think the game is more important than the demo. Nevertheless, I am interested to know how the demo works on real console.

James-F2, 2019-09-28:
I think a lot more people are familiar with the Bad Apple demo than Magical Drop.
I have 5 SNES consoles, I will test the Bad Apple demo and report back.

James-F2, 2019-09-28:
On real hardware the demo audio boots consistently, BUT, audio sometimes stops in the middle of the demo while video continues running.
Also, I can see a flickering scanline on the bottom left corner with the hardware which does not appear in the core or bsnes.

srg320, 2019-09-28:
In the core the sound also sometimes stopped somewhere in the middle of the video, so we slightly adjusted DSP frequency because of this demo (but this is not entirely correct). Audio stopping happens in the different spots because the frequency of both quartz resonators (21MHz and 24MHz) is slightly different on each console.

MP2E, 2019-10-06:
This appears to be fixed as of #148

James-F2, 2019-10-06:
Fixed.

closed 2019-10-06 by James-F2