MiSTer-devel/SNES_MiSTer #157: SDD1 ROM access

gyurco, 2019-10-17:
Just another question about the SDD1 chip: I see the decoder and the CPU accesses the ROM in a time-shared manner. In the worst case, 3 master clocks/chip. It seems too fast for me (140ns). Is it how the real console does it? AFAIK the "Fast" ROMs were accessed in 6 master clocks normally, this is double rate.

srg320, 2019-10-18:
Yes, SDD1 and SA1 use time-multiplex acces the ROM (16 bit 120ns). Both chips have a separate bus for ROM, CS/OE signals either not present or active all the time.
Some info https://forums.nesdev.com/viewtopic.php?f=12&t=17538

gyurco, 2019-10-18:
That's a really fast access then. Now I only wonder why the core expects the data in 2 clock ticks only (<100ns), in InputMgr, WAIT_ACCESS = 1. It's even a bit of challenge for the sdram controller in MiSTer, too.

srg320, 2019-10-18:
Original SDD1 has fifo, but MiSTer sdram controller "have time" to out data to InputMgr without the need for fifo.

gyurco, 2019-10-18:
I don't speak about the data stream speed, but the sequence is: assert rd_n, set wait_access to 0, set wait access to 1 and write req to 1 in the same time, that means there are only 2 cycles to the SDRAM controller for returning the data.
I see first it was done differently, then there was a try with WRITE_RQ <= '1'; instead of :=1, but it wasn't used.

closed 2020-01-22 by sorgelig