MiSTer-devel/SNES_MiSTer #159: Tales of Phantasia glitchy menu graphics

softtest03, 2019-10-25:
Load the save file. Open up the game's menu (SNES button X). Click on "items" ("アイテム"). The graphics behind the item list's text will glitch as you move around with the D-Pad.

Save file: https://ufile.io/z8cnnyu5

I'm playing with an unmodified Japanese rom.

srg320, 2019-10-27:
On bsnes, snes9x and no$sns I see the same thing. Сan you show how it looks on real SNES?

MP2E, 2019-10-27:
It does the exact same thing on my 1chip SNES, I don't think this is a MiSTer bug

softtest03, 2019-10-27:
Sorry, I don't have an SNES to confirm this but I will assume that you are right and it's a false positive.

MP2E, 2019-10-28:
No problem! I think it's always good to have someone double check if behavior seems weird.

closed 2019-10-28 by sorgelig