MiSTer-devel/SNES_MiSTer #164: More IRQ & NMI tests from higan

James-F2, 2019-11-08:
https://gitlab.com/higan/snes-test-roms/tree/master/KungFuFurby-test-ROMs

James-F2, 2020-01-06:
including some previous tests from here: https://github.com/MiSTer-devel/SNES_MiSTer/issues/105

Real console GPM-02 results;

demo_irqtest. (pass 19/20). <--- MiSTer **Fails** consistently
demo_nmitest (pass 20/20).
**irq (pass 0/5) real console fails consistently.**
**nmi (pass 0/5) real console fails consistently.**
test_dma (20/20)
test_dmasync (20/20)
test_dmavalid (20/20)
**test_hdma (pass 10/20,  fail: 10)** - (MiSTer passes consistently)
**test_hdmasync (pass 4/10, fail 6)** - (MiSTer passes consistently)
**test_timing (pass 5/20, fail 15)** - (MiSTer passes consistently)
**test_irq (pass 8/20, fail 12)** - (MiSTer passes consistently)
test_irq4200 (pass 20/20)
test_irq4209 (pass 20/20)
test_irqb (pass 20/20)
test_mdrhdma (pass 20/20)
**test_nmi (pass 7/20, fail 13)** - (MiSTer passes consistently)

So MiSTer is too perfect now, except first IRQ test as for release 20200106. :)

birdybro, 2022-05-22:
>test_nmi (pass 7/20, fail 13) - (MiSTer passes consistently)

Just tested and this fails now on MiSTer consistently. Tracing the regression by testing different releases and... it began failing on release 20201027. The key as to why the test no longer passes is potentially within these two commits

https://github.com/MiSTer-devel/SNES_MiSTer/commit/79e55d6a3ad7cce9eec5786f40bc9483f6741400

https://github.com/MiSTer-devel/SNES_MiSTer/commit/8e53ff1b556adda5fd42a1ea95fb711d97921104

I agree, that the 3rd and 4th tests are bad tests if they fail consistently on a real console.