MiSTer-devel/SNES_MiSTer #174: Super Mario World Cape Powerup Sound Too Slow

daroga, 2020-01-02:
Hello!

This is my first time posting an issue. Forgive me if I mess things up.

I just played through all of Super Mario World. I noticed the sound effect that plays when you get the feather power up plays at about half speed (compare to the sound effect here on real hardware at about 10:53: https://youtu.be/0CcgWb4GhEA?t=650). I didn't notice any other issues, sound or otherwise, throughout the game.

Thanks!

James-F2, 2020-01-05:
Here mister sounds exactly like on the real console and this video.

James-F2, 2020-01-06:
@sorgelig 
If OP does not reply shortly, this can be closed.

daroga, 2020-01-06:
I'll chalk it up to an odd issue on my end. If no one else is having a problem, let's call it closed.

Thanks, everyone, for all of your work!

closed 2020-01-06 by daroga