MiSTer-devel/SNES_MiSTer #175: Super Metroid & A Link to the Past Combo Randomizer Beta hangs when transitioning

TomFaulkner, 2020-01-05:
This is not a duplicate of #81, however the problem is the same. On the beta at https://beta.samus.link a rom can be generated.

When transitioning from Metroid to Zelda the screen goes black and hangs. The core is still able to  get to menus and such, but the game never moves.

To transition between games:

> To transition into Zelda, go left from Samus' ship in the beginning. Then through the Blue Door, and in the next room, go left until you reach a wall. Go down from here until you encounter a blue door on your right hand side. Go through this door, and it will transition into Zelda

Thanks for all of your hard work.

MP2E, 2020-01-05:
Can confirm this is an issue. very strange..

MP2E, 2020-01-05:
I looked at the ROM source repository for this randomizer, and a change stuck out to me. transition code used to be saved at B8/F8, now it is saved at AA/EA

https://github.com/tewtal/alttp_sm_combo_randomizer_rom/commit/d4898009fdccd110a568dc5d5081b611b78a2431#diff-c90efad2b4de8d52423e0005b4e1733f

sorgelig, 2020-01-06:
I don't see anything strange in it. Hacks and mods are often use edge cases to access the HW, mostly undocumented.

James-F2, 2020-01-06:
@TomFaulkner 
Does your generated rom freezes on the sd2snes?

TomFaulkner, 2020-01-06:
I don't have an sd2snes, but others have been using the generated roms without issue. I can ask someone on the smz3 discord to try the rom tonight.

Also, @tewtal did mention that he increased the sram size with this version.

srg320, 2020-01-06:
Can anyone share a rom?

MP2E, 2020-01-06:
I sent it to you via the Atari Forums in a private message, thanks for looking into it!

James-F2, 2020-01-06:
Works fine on SD2SNES.

Xenobond, 2020-01-06:
Just tested with latest release version (20200106) and it is working fine for me. Tested both with a rom generated back in October and one created today.

James-F2, 2020-01-06:
It is only that **specific** seed in the first post link that freezes.

TomFaulkner, 2020-01-06:
@James-F2 , I tried two seeds, both have the same issue.

MP2E, 2020-01-06:
It's not a specific seed that causes it, any beta seed, which has a newer base rom, has the issue.

srg320, 2020-01-07:
When transitioning occurs, this code is executed:
F8006D: LDA   $A17882
F80071: CMP  $A17880
F80075: BEQ   $0096
But neither A17880 nor A17882 are initialized until this point. Most emulators fill backup ram with 00 or FF before starting the game and this code works correctly. On real hardware the memory contains random values for the first time and therefore must first be initialized in the game code.

MP2E, 2020-01-07:
Thanks for the fix! I'm surprised the sd2snes works. Perhaps it fills backup ram as well. I'll ask in the combo randomizer discord about this

James-F2, 2020-01-07:
from ikari (sd2snes dev):

> it should be FF. I know 00 breaks some things and FF breaks some other things.

Kreeblah, 2020-01-07:
If this is going to be added, could it be added as an option?  Maybe default to uninitialized, but have options to initialize as 00 or initialize as FF?

sorgelig, 2020-01-07:
So obviously it's game mod bug rather core's
It should be reported to modder.

TomFaulkner, 2020-01-07:
> So obviously it's game mod bug rather core's
> It should be reported to modder.

It had been. I'll close this, since it has been shown that it is not a MiSTer issue. Thanks for the work in identifying and documenting the problem.

closed 2020-01-07 by TomFaulkner

James-F2, 2020-01-07:
Question is,, is there any game that will be negatively affected by this FF change?
I don't think any games needs random values in the backup ram,, if there was, it would have been reported on sd2snes,, unless,, the sd2snes has quirks for these games.

MP2E, 2020-01-07:
I don't think there are any games that would be negatively affected by filling the save ram with FF, so I think this fix is good.

However, if it is discovered that random backup save RAM is necessary, I'm ok with this change being reverted. I reported this to the combo randomizer developers and one of them took a deeper look and he said this exposed 2 bugs that he would like to fix soon, one of them being that the save memory should be initialized by the hack.

Kreeblah, 2020-01-07:
While there might not be any games that are negatively impacted, there is still value in attempting to be a reference platform, as people running into bugs when running code on MiSTer are more likely to look into why they happen and fix them so things run on real hardware.  In this particular example, they've had a bug open for their ROM hack for a year and a half with it not working properly on a Super Everdrive.  The behavior was basically identical to what's reported here, so I'd be surprised if it's not the same issue.  But, it didn't get fixed since nobody had a Super Everdrive (and, with the SD2SNES pre-filling save RAM, probably assuming that it was fine on real hardware), and assumed it was an issue with that particular cart.

Having more independent implementations (especially ones attempting for as much accuracy as possible) available to highlight issues that deviate from actual behavior is a good thing, as it helps to highlight when issues are because of implementation quirks.

TomFaulkner, 2020-01-07:
I don't disagree with @Kreeblah's conclusion, but this issue just came up, and while it may be similar, it does not exist with the current release version of the randomizer.

While it would be great if this just worked right now, it would be better to fix the ROM rather than rig the hardware. Especially given that this randomizer ROM is in active development. If it was a ROM from an unreleased game in the 90s and it couldn't be patched I think that might lead to an option in the menu. But I have not contributed to MiSTer so my opinions shouldn't have much weight.

srg320, 2020-01-08:
> Question is,, is there any game that will be negatively affected by this FF change?

I only know about one game with a bsram issue #20, it happens if clear the bsram with 00.