MiSTer-devel/SNES_MiSTer #183: Mode 7 in Illusion of Gaia not working right

PupilEX, 2020-01-08:
For some reason, the SNES core on MiSTer has trouble with Illusion of Gaia during the mode 7 world map sequences. Originally the screen would go black for the entire mode 7 sequence. After the update the screen now just goes dim during the mode 7 sequence and flickers and it's fine once the sequence is over. Mode 7 works fine on every other game that I tested, only Illusion of Gaia seems to do it, at least from what I've found.

James-F2, 2020-01-08:
I confirm.

Exit school and got up to castle.
Save file:
[Illusion of Gaia (USA).zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/4036857/Illusion.of.Gaia.USA.zip)

srg320, 2020-01-08:
Thanks. It's good that the bug is old.

James-F2, 2020-01-22:
@sorgelig 
Can be closed.

closed 2020-01-22 by sorgelig

vanreece, 2020-02-08:
I can confirm that this is still happening on SNES_20200108.rbf

@James-F2 did you think that this wouldn't happen any more on Illusion of Gaia?

vanreece, 2020-02-08:
Well, actually, it seems to work fine after the first mode 7 event without issue? Not sure what to make of that.

James-F2, 2020-02-09:
It has been fixed but not released yet.
Please don't waste the developrs time bumping closed issus if you dont know how github works, there is a forum for that.

vanreece, 2020-02-09:
Fair enough, sorry to waste time. Was just trying to be helpful. I'll try to find the forum next time.