MiSTer-devel/SNES_MiSTer #184: Emudetect, scrambled screen.

James-F2, 2020-01-10:


paulb-nl, 2020-01-10:
Source links:
[https://www.smwcentral.net/?p=files&u=1686](https://www.smwcentral.net/?p=files&u=1686)
[https://bin.smwcentral.net/u/1686/emudetect21_imgsrc.zip](https://bin.smwcentral.net/u/1686/emudetect21_imgsrc.zip)
[https://bin.smwcentral.net/u/1686/emudetect21_src.zip](https://bin.smwcentral.net/u/1686/emudetect21_src.zip)

James-F2, 2020-01-11:
Works after "Reset".
Maybe do a reset immediately after every rom load?

James-F2, 2020-01-11:
Interesting.
Running Emudetectafter First Drop (https://github.com/MiSTer-devel/SNES_MiSTer/issues/185) (which doesn't work yet) will results in puppy image.
So there is some interaction between these two.

srg320, 2020-01-11:
Try with a257e51

James-F2, 2020-01-11:
I'm testing with ""WRAM: fill with FF"" for the last 40 minutes.
I get same thing as in third post.

paulb-nl, 2020-01-11:
Release 0106 and earlier show the puppy normally at cold boot. 

I have also tried it with an old SuperFlash cart on a 1-CHIP and it shows the real hardware image from above. Though that cart also has a (simple) menu.

Looking at the source the numbers for real hardware should be:
$69,$69,$00,$02,$FF,$00,$02,$0E,$26,$54,$39,$FF,$00,$00,$00,$00

If I am not mistaken it looks like the first 2 are from the first untouched WRAM test.

```
;INITIAL WRAM VALUES
;This test requires a block of untouched RAM.
TestWRAM:
LDA #$00
TAX
XBA
LDA #$FF
.Loop
AND $0200,x
XBA
ORA $0200,x
XBA
INX
BNE .Loop
%store(a)
XBA
%store(a)
```

0106:
[image]

0108:
[image]

paulb-nl, 2020-01-11:
This rom shows the real hardware screen at cold boot after I hacked the rom to pass the second CGRAM test. It needs a result of FF there and it is currently 00. CGRAM probably needs randomization.

> If I reload the rom again (making it second rom), then reset, it will show "real hardware",

Yes the rom itself has now written to CGRAM so it passes now. The corrupted screen at cold boot is only with release 0108.

```
;ANOTHER CGRAM TEST
;This test does not work if anything has written to CGRAM before this code is run.
TestCGRAMORA:
LDA #$00
TAX
.Loop
STX $2121
LDY $213B
ORA $213B
INX
BNE .Loop
%store(a)

```

James-F2, 2020-01-12:
Only two significant commit between 0106 and 0108.
Probably: SCPU: fix interrupts,, but that commit fixed a lot, so need to be careful if changing something in it. https://github.com/MiSTer-devel/SNES_MiSTer/pull/180
Whatever it does, Reset fixes the corruption.

paulb-nl, 2020-01-14:
Release 0108 with SCPU: fix interrupts reverted fixes the corrupt screen at cold boot so there is still some inaccuracy there.

James-F2, 2020-01-16:
After the latest commit (https://github.com/MiSTer-devel/SNES_MiSTer/commit/94610ab4b1b75e5859a140e4347e904d417d2263), the test fails consistently.
I will compile a 0x55 wram fill and test.

James-F2, 2020-01-16:
Here is what I get with 0x55 and "Fix RAM clearing skipping uneven bytes". :laughing: 
[image]

paulb-nl, 2020-01-16:
😄 Here is a commit for choosing between 4 different RAM patterns: https://github.com/paulb-nl/SNES_MiSTer/commit/d97cd6b49a59c207dc29e83b440fae451f2b8eb1

James-F2, 2020-01-16:
hmm
To tell you the truth, I think this will complicate user experience.
Is there are more 'automatic' way to simulate real snes ram ranadomness?

paulb-nl, 2020-01-16:
It is just for testing the different fill patterns. 

I have done some testing on my consoles with the Mario World jailbreak that has a hex viewer in it.

At cold boot 2x 1-CHIP and an APU-01 mostly have a 00 FF pattern. An older SNES has a 44 77 11 DD pattern. It is the same at every boot.

The first 2 patterns in my commit are both detected as real consoles so maybe we could make a choice of just Console 1 & 2.

James-F2, 2020-01-16:
Yeah.
Two "Initial WRAM" options clearly labeled as "1Chip" and "3Chip" will be great.

What 'older' 3Chip do you have, CPU, GPM or RGB? 
I will test on my consoles, if I can find the rom.

paulb-nl, 2020-01-16:
That's not entirely accurate because my APU-01 has the same 00FF pattern as the 1-CHIPS.

I haven't opened the older SNES yet but based on the video output it is probably a CPU-01. It is a PAL one.

I put the jailbreak https://github.com/MiSTer-devel/SNES_MiSTer/issues/87 on my original cart by overwriting its save. If you run it on the SD2SNES then it will just fill WRAM with 55 before starting the rom.

The original cart has some untouched RAM at 7F88-7FC7

James-F2, 2020-01-16:
ah you're right.
I wonder if the 44 77 11 DD is exclusive to oldest CPU-01 or pal consoles.
00 FF pattern is a good start though.

paulb-nl, 2020-01-17:
- The Anthrox demo has some scrolling artifacts beneath the letters with 9966 and 55 and works fine with 00FF and FF.
- Emudetect shows real hardware only with 9966 and 00FF.
- TricksAsm does not work with 00FF.

I think its best to add the "Initial WRAM" setting with "Type 1" (9966) and "Type 2"(00FF). What do you think?

James-F2, 2020-01-17:
Does the 1Chip change the pattern to FF 00 at some point or stays 00 FF?
According to this image you posted earlier the pattern changes from FF00 to FF00,, can you verify with the SMW Jailbreak cartridge?
[image]

I don't see any scrolling artifact in Anthrax - Legendary Intro demo with 0x55, where exactly?
Emudetect displays "real hardware" on sd2snes which is 0x55, and with my build too.
It doesn't matter if TricksAsm doesn't work, this demo was created on emulators and was never meant to work on real console,, we strive to be as close to the real console as possible.

I would go only for 00FF for simplicity, this is a waste of menu option imo.
Can you check again if the console changes from 00FF to FF00 every 64 bits like the image above?

James-F2, 2020-01-17:
I could be wrong.
Maybe several initial wram options will be useful since people play hacks made for emulators?

paulb-nl, 2020-01-17:
Emudetect shows the Bsnes accurate screen with 55 right? so it is not really pass.

Yes the consoles with 00FF change to FF00 every 512 bytes. This is already implemented in my commit. The console with 44 77 changes every 256 bytes which is also implemented.

Both 9966 and 00FF are accurate to console so some demos will not work on some consoles. Will we emulate only 1 type of console?

Can you verify with the Mario World jailbreak that 7F88-7FC7 contains 55 only with SD2SNES?

James-F2, 2020-01-17:
Yes full of 55 in 7F88-7FC7 addresses on the sd2snes.
You are right, probably best to have 9966 and 00FF to cover all possibilities of real consoles.
Maybe even keep the 55 and FF options for maximum compatibility with hacks?

Optiroc, 2020-01-17:
I really must question the usefulness of having this as an option. Will anything except rather shoddy homebrew behave differently with 0x00FF and 0x9966 patterns? Adding emulator quirks seems like a really slippery slope to me. Patch the bad software, not the core.

paulb-nl, 2020-01-17:
Super Offroad might. It has a random number generator which reads from uninitialized ram. The 00FF, and 9966 patterns come from real SNES consoles. Apparently 55 is a pattern with real consoles too but I haven't encountered that one myself. 55 is the one having issues with Super Offroad.

Just initializing with random garbage is not accurate to real consoles.

Kreeblah, 2020-01-17:
I'm not a fan of adding emulator quirks either, but if there is support for them, it would at least be nice if they were something that users have to opt into.  It'd be fantastic to see MiSTer be, effectively, a reference platform.

@paulb-nl - Would you have any thoughts regarding adding in functionality to these options to address the BSRAM change in https://github.com/MiSTer-devel/SNES_MiSTer/commit/a7c06769ba75faf3c87c9721c5f6d9c53826186c?  That was a similar situation where a change was made to accommodate a ROM hack that didn't initialize RAM.

James-F2, 2020-01-17:
First pattern in that image looks like this:

0x9B99,999B,999B,9B99
0x6666,6666,6666,6666
0xE666,66E6,66E6,E666
0x9999,9999,9999,9999

so it's not exactly 99 66.

paulb-nl, 2020-01-17:
Currently it works like this:
0x9999, 9999, 6666, 6666
0x6666, 6666, 9999, 9999

inverts every 256 bytes:
0x6666, 6666, 9999, 9999
0x9999, 9999, 6666, 6666

I have dumped some saves from my cartridges to see what the initial save ram content is. 

A Super Metroid cart was mostly filled with 0x54 mixed with some 0xFF.

A PAL Mario World cart had mostly 0x54 mixed with some random values duplicated 4 times like this 0xADADADAD 1B1B1B1B. 

A US Mario World cart had 0x00FF and 0xFF FF FF FF 00 00 00 00 patterns.

I think it is probably fine to leave the BSRAM at 0xFF fill? Or alternating 0x54 and 0xFF patterns.

James-F2, 2020-01-18:
BSRAM at 0xFF is fine, since it doesn't have any incompatibility with games.
-
Anthrox - Rotating Mode 7 demo is scrambled with 00FF wram pattern.
These demos are meant for emulators, and after discussing this for several days I think it would be best to just decide on one value that matches the real console like 00FF and ignore some demo's incompatibility with real hardware.

paulb-nl, 2020-01-18:
I want at least to have the option of the 2 SNES patterns. The Anthrox demos were meant for floppy copiers not emulators.

I don't like the randomness of Bsnes because things will break randomly depending on the pattern.

I will send a pull request with the options "00FF(SNES1),9966(SNES2)" so that 00FF is the default. "00FF Real Console" is too long for the menu.

James-F2, 2020-01-18:
Might as well include 55 and FF if the option already exists.

Do you think it is safe to revert this?
I don't experience any problem.
https://github.com/MiSTer-devel/SNES_MiSTer/blob/362b3e1943308c1a96f3cb671272eaff588b4a23/src/65C816/P65C816.vhd#L484

srg320, 2020-01-18:
> Might as well include 55 and FF if the option already exists.
> 
> Do you think it is safe to revert this?
> I don't experience any problem.
> https://github.com/MiSTer-devel/SNES_MiSTer/blob/362b3e1943308c1a96f3cb671272eaff588b4a23/src/65C816/P65C816.vhd#L484

Check issue with blue background in Battletoads intro.

paulb-nl, 2020-01-18:
> Check issue with blue background in Battletoads intro.

Yes Battletoads has an issue where the background is blue instead of black with that change. If I understand correctly Battletoads needs NMI to start immediately after the DMA transfer is finished after Vblank has passed during DMA.

I have Mesen-S set to fill RAM with 0xFF. I see with The First drop demo in Mesen that it does a very long (because of the 0xFF) DMA transfer inside the NMI handler and after the transfer another NMI starts while still inside the NMI handler.

Battletoads, First drop demo and Emudetect all work properly in Mesen-S.

James-F2, 2020-01-20:
Reverting the **if CE = '1' then** line fixed this issue: https://github.com/MiSTer-devel/SNES_MiSTer/issues/196

James-F2, 2020-01-22:
CE = '1' was fixed by srg320.
An option to select WRAM added by paulb-nl.
Closing.

closed 2020-01-22 by James-F2