MiSTer-devel/SNES_MiSTer #185: First Drop demo,  black screen

James-F2, 2020-01-10:
http://www.pouet.net/prod.php?which=12936

paulb-nl, 2020-01-14:
Black screen on release 0108 but works on release 0106 or 0108 with SCPU: fix interrupts reverted.

James-F2, 2020-01-17:
Works with 00FF and "SCPU: fix interrupts".
Apparently there is some interaction between this SCPU fix and WRAM.

@srg320 Can you look why "SCPU: fix interrupts" prevents this First Drop demo from starting and corrupting the Emudetect screen?

paulb-nl, 2020-01-17:
I have narrowed it down to this: https://github.com/MiSTer-devel/SNES_MiSTer/blob/362b3e1943308c1a96f3cb671272eaff588b4a23/src/65C816/P65C816.vhd#L484

Changing this back to "if RDY_IN = '1' and CE = '1'" fixes this demo and Emudetect corrupt screen.

It looks more related to the Vblank NMI because First Drop does not use HV interrupts.

James-F2, 2020-01-17:
After 20200106 Bad Apple and 2.68MHz demo's were corrupted, and "SCPU: fix interrupts" fixed them.
Can you check if you change has any effect on these two?

paulb-nl, 2020-01-17:
Those two seem to work fine with that change.

James-F2, 2020-01-17:
I compiled a version with the four wram option and this line reverted,, everything works correctly.
I assume this is safe to revert.

James-F2, 2020-01-22:
CE = '1' was fixed.
An option to select WRAM added by paulb-nl.
Closing.

closed 2020-01-22 by James-F2