MiSTer-devel/SNES_MiSTer #187: Snes9x demo v1.16 by TricksASM, black screen

James-F2, 2020-01-10:
http://www.pouet.net/prod.php?which=69308

James-F2, 2020-01-11:
Start working after **several** "Reset"s.

James-F2, 2020-01-11:
@srg320 
Seems to be fixed with ""WRAM: fill with FF"".
Works when loading the rom,, when Reset I get black screen,, but same thing on sd2snes.

srg320, 2020-01-11:
This and other 2 demos are another example of bad programming.

James-F2, 2020-01-11:
Definitely.
First Drop demo will not even start, and Emudetect has false detection.
Maybe sd2snes doing something?
Maybe sd2snes reset when loading rom so that fixes Emudetect?
Demos usually run on emulators and sd2snes,, nobody is going to make real cartridge for demo only.
We don't know if emulators or sd2snes have special quirks for these demos.

srg320, 2020-01-11:
Of cours, sd2snes first launch its rom, which clean and initialize everything and then load rom after select game.

James-F2, 2020-01-11:
I've asked @mrehkopf (ikari_01) on discord, maybe he can give exact answer what the sd2snes does.

mrehkopf, 2020-01-12:
sd2snes has a procedure to "de-init" much of the things that the menu used before running a game.
- fill all APU RAM $0100-$03ff with $aa, $0400-$ffff with $00 (1. to please buggy sound routines like Soul Blader; 2. to avoid echo artifacts right after sound init on some games)
- stop HDMA
- turn off screen
- fill all WRAM with $55
- init all DMA registers with $ff
VRAM is left alone.

James-F2, 2020-01-12:
Thank you mrehkopf, that helps a lot.

paulb-nl, 2020-01-12:
It is stated that $55 is not good for Super Off Road.
[https://octolinker-demo.now.sh/TASVideos/BizHawk/issues/1701](https://octolinker-demo.now.sh/TASVideos/BizHawk/issues/1701)

James-F2, 2020-01-16:
This issue is fixed with "WRAM: fill with FF", but I'm keeping it open till we decide what to do with wram randomness at boot.

James-F2, 2020-01-17:
> A description of Super Off Road's usage of uninitialised RAM for its random number generator 

"uninitialized"  meaning 00FF?

paulb-nl, 2020-01-17:
It can be 00FF or another pattern depending on the ram chips in the consoles as we've seen. 

Yes WRAM resets if you leave it off long enough otherwise if you quickly change the cartridge it can still retain some of the written bits.

James-F2, 2020-01-22:
An option to select WRAM added by paulb-nl.
Closing.

closed 2020-01-22 by James-F2